winder/Universal-G-Code-Sender #1171: Command line interface

breiler, 2019-01-11:
## Feature request
Some people has requested a CLI interface for using Universal Gcode Sender from terminal (#945, #278, #882). I've started playing around with such an interface and are wondering if this still could be of interest?

These are the features right now:
* Set baud rate to connect with
* Lists all available ports
* Choose what type of controller firmware we are connecting to, defaults to "GRBL". Available firmwares: [GRBL, TinyG, Testing (Delay), Smoothie Board, Testing]
* Opens a file for streaming to controller and will exit upon completion.
* Home machine and will be done before any gcode files are sent to the controller.
* Select which port for the controller to connect to. I.e /dev/ttyUSB0 (on Unix-like systems) or COM4 (on windows).
* Resets alarm

Different output types can be selected: a plain stream of the commands sent to the controller or a progress bar. Here's an example of progress bar:
[image]

I'm thinking that we can package a separate start script together with either UGS classic or it's own package. What do you think?

Check out the current implementation (still a work in progress):
https://github.com/breiler/Universal-G-Code-Sender/tree/cli/ugs-cli

carneeki, 2019-01-12:
I can see value in this, particularly for automating batch jobs.

To that end, I can also see value in a "--no-home" override.
Use case 0: I'm doing a number of the same g-code file after changing out
pallets of work.
Use case 1: I'm sending a series of files, each file after a tool change.

Both use cases are almost the same, and won't necessarily require
re-homing. There are probably other cases that I haven't thought about too.

(actually, considering my workflow, I'm almost inclined to suggest making
"--no-home" the default, and add a "--home" option to do a homing cycle
before sending, or have the default saved in the configuration file)

closed 2019-04-01 by breiler

okemawo, 2021-08-13:
> ## Feature request
> Some people has requested a CLI interface for using Universal Gcode Sender from terminal (#945, #278, #882). I've started playing around with such an interface and are wondering [...]

Link is dead please post a new one

breiler, 2021-08-13:
@okemawo: https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-cli

zzaacchh, 2022-09-22:
I am interested in this functionality!  Thanks, @breiler for developing it.  Am I right in understanding that this has been merged into the mainline UGS source tree and binary?  I see instructions for running the CLI via the ugs-cli-2.0-SNAPSHOT.jar jar file, but I don't see that jar file available for download.

Any help will be greatly appreciated.  I feel like I'm probably missing something obvious.

breiler, 2022-09-22:
@zzaacchh this is not obvious so I think you have the right to be confused. :smile: 

It is available in all current releases listed here:
https://github.com/winder/Universal-G-Code-Sender#downloads

You will need the UGS Classic edition. There is only a start script for Unix-based systems, but you should be able to run it using the following:

```
java -Xmx256m -cp UniversalGcodeSender.jar com.willwinder.ugs.cli.TerminalClient -h
```

zzaacchh, 2022-09-25:
@breiler  It worked!  Thank you.   Is it likely that these features will one day make it into the full UGS release?

breiler, 2024-07-23:
@lurueger make sure that you are using the latest version 2.1.8 and that nothing else is connected to device at the same time.

jaganathsr, 2024-11-10:
I am facing the same issue. There are no other apps connected to the device.