winder/Universal-G-Code-Sender #1173: Soft reset via web-Pendant UI

aleslukek, 2019-01-14:
## Feature request
I am working on a hardware external control, that works via Pendant web ui. It is sort of a hack. I had lots of trouble when connecting buttons directly via GPIO to my raspberry pi to act as keyboard.
So I decided to make an ESP8266 module that has buttons connected and interfaces with ugs via Pendant UI. It also has a LCD to show data.

I got most of it working. Now I can have more commands with combination of buttons. Homing, reseting zero, resetting z zero, quick jog, slow jog, probe, clear alarm, switch to laser mode, show just a laser dot (low power), start job, pause, resume, cancel... Only thing I cannot do is soft reset after hard limits are triggered. Soft reset works if grbl is idle and I send $C twice. But not when limits are triggered. I was thinking of a physical connection to arduino reset button. But I would be risking that if ESP resets for some reason (ESP sometimes does that) it would send low signal to arduino that would reset it. If this happened when arduino is working it could be bad. So this is out of the question. If there was a get request for soft reset I could hard code that if cnc is working there would be no way of sending reset request.

I tried ^X, ctrl-x, 0x18 , $C twice, .... nothing works. https://github.com/winder/Universal-G-Code-Sender/blob/0f1465381c1faac1ec926f13956a15d5c19798b9/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L60 

For some reason UGS seems to block those commands, or doesn't translate them to arduino properly.
I interface UGS with GET request. I have to have pendant ui enabled (auto enable on startup would be awesome!). 
cncip:8080/sendGcode/?gCode=xxx
so xxx is the gcode I want to send. Some commands are in text like "RETURN_TO_ZERO", "CANCEL_FILE", "SEND_FILE".  https://github.com/winder/Universal-G-Code-Sender/blob/0f1465381c1faac1ec926f13956a15d5c19798b9/ugs-core/src/com/willwinder/universalgcodesender/pendantui/PendantUI.java#L158-L216

It would be great if there was "SOFT_RESET" as well. I tried a lot, also got through UGS source code, but there was no hidden soft reset get request.

tldr: get request - button for soft reset via pendant web ui and ability for pendant web ui autostart on startup.

AbySet, 2019-01-17:
I'm not java programer but i though this may help a little bit:
Look for
          issueSoftReset()
to add this
case "something":
          mainWindow.issueSoftReset();

SOFT_RESET variable already used for machine status reading.

aleslukek, 2019-01-17:
Was looking into it. Hopefully this could work, however things usually don't go this smoothly for me. I am however not intimately familiar with compiling java. Will read into it and will try it.

Concerning an option in menu for auto-enable pendant it is truthfully beyond my java skills.

aleslukek, 2019-01-17:
I am working on a esp8266 module that has bunch of buttons and an LDC connected to it to control via UGS. Code is mostly done, have to test it a bit more, and will put it on GitHub once it's done. So far it is working great with some small issues of course.

Soft reset is a last feature I'd like to implement.

AbySet, 2019-01-17:
Ah ok sounds good, i'll be waiting for it, i've ESP32 but not using it for now.
Good job there.

aleslukek, 2019-01-29:
Truth be told, haven't got around to java recompiling ugs and enabling soft reset via pendant ui. However I have mostly finished my UGS Wifi hardware pendant code and hardware and must say it works great!
Code and schematic is available here https://github.com/aleslukek/UGS-Wifi-Pendant
This works only on UGS and i only tested it on grbl 1.1f.

Will try to get back to this soft reset issue again, hopefully soon!

breiler, 2019-04-11:
Have a look at the new API:
https://github.com/winder/Universal-G-Code-Sender/blob/6ffab3e26b04bffa7bf79d2b9c7d1ab88ef2f893/ugs-pendant/src/main/java/com/willwinder/universalgcodesender/pendantui/v1/resources/MachineResource.java#L173

breiler, 2019-04-24:
A new build is out with the new API which now has a method for softReset.

closed 2019-04-24 by breiler