winder/Universal-G-Code-Sender #1176: port busy

mgorman0178, 2019-01-20:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

mgorman0178, 2019-01-20:
Hi my machine stopped halfway thought what i was doing! I checked my software and an error message come up saying Error opening connection (jssc.serialportexception): Port name- COM3; Method name-openport(); Exception type- port busy!? I turned it of and on but still same message! What do i do? Oh and the visualize don't work aswell???

breiler, 2019-04-26:
Closing this as no additional information was provided.

closed 2019-04-26 by breiler

PauCatala, 2023-02-12:
Hi, I have the same problem, and I cannot fix it. I have nothing connected to my PC except for UGS. I have my Arduino connected to COM8, and even if I connect it to other ports the same error appears. Is there anything I can do?

breiler, 2023-02-12:
@PauCatala please create a new issue.