winder/Universal-G-Code-Sender #1178: probe on double tap not functioning in inch mode

jwhelget, 2019-01-21:
UGCS 2.00 platform nightly build Dec 18 2018
windows 10
GRBL 1.1
probe thickness .12"
probe distance 1.00"
slow probe 5
fast 10
retract .59"

After the first contact it retracts what would need to be something close to 1.00" since the probe distance needs to be 1.00" or it will alarm out.   Not sure what the retract amount should be but what I have seen on a demo is likely .10" or less.  After the slow probe it retracts to .59.   Even if the retract is .1" it retracts enough to need the 1.00" probe distance or will alarm out.   Works in metric .....sorry I need to work in inch.   

## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

AbySet, 2019-01-21:
Using inchs:
probe thickness .12"
probe distance 1.00"
slow probe 5
fast 10
retract **.59"**
After testing with same settings, here is what i've got:
- After [Fast find] motion it retracts to **1 inch.**
- After [Slow measure] motion it retracts to 0.59 inch.

Using mm:
probe thickness 3 mm
probe distance -25.4 mm
slow probe 100
fast 200
retract **14.986** 
After testing with same measurments converted to mm, here is what i've got:
- After [Fast find] motion it retracts to about 1 or 2 mm.
- After [Slow measure] motion it retracts to **10 mm.**

That's all i can say, i have to say i'm confused with this because there is one retract parameter where there is two retract motions when probing, so is the parameter for the first one or the second one?
But i think i may be for the second one cz it's the one that matters.

So two questions remains:
- Why the first retract is 1 inch (using inch) and is about 2 mm (using mm)?
- Why the second retract is correct 0.59 inch (using inch) and is 10mm when it's set to 14.986 (using mm)?

jwhelget, 2019-01-21:
You are now at least the third person now to confirm odd behavior for this function.

jonadams1010, 2019-01-29:
I submitted a change yesterday which will use the "Retract Amount" setting for mm and inches. It will use this only for the first retract (between the fast and slow probes). After the slow probe it will return to the start position, which is consistent with the XY and XYZ probes.

I don't know if the nightly build has picked it up yet, but if it has, you could give it a try.

AbySet, 2019-01-29:
Thank you
I'll try that as soon as possible after next build.

winder, 2019-01-29:
I still haven't setup a new nightly build process, the only way to get this change right now is to download the source

AbySet, 2019-01-29:
I prefer to wait cz i'm newbe at java, thank you Winder.

breiler, 2024-02-07:
Closing as this is no longer a problem in later versions of UGS. Tested on version 2.1.4.

closed 2024-02-07 by breiler