winder/Universal-G-Code-Sender #1181: When send my gcode file an error was detected while sending '0S30' error

jashwant1234, 2019-01-24:
[image]

## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

winder, 2019-01-24:
Looks like you have a typo in your gcode, try removing that line

jashwant1234, 2019-01-24:
Plzz help me to make gcode in from of .nc

winder, 2019-01-24:
The screenshot tells you what line didn't work. Try opening the text file and removing that line. If you still have issues you'll need to configure your CAM software for GRBL.

jashwant1234, 2019-01-24:
How i configure i don't know

AbySet, 2019-01-25:
Tell us where did you get that file from so we can help, if that file is generated by some software you are using, we need to know what is it, if not then just delete that line.

AbySet, 2019-01-26:
Hey;
I forget to mention: In your case the job is just paused not canceled, just press RUN button again and the machine will skip that command and continu. But if the file has many of those commands, they will be disturbing and you better delete them before you run the job.

breiler, 2019-10-09:
There were no follow ups from user, closing issue.

closed 2019-10-09 by breiler