winder/Universal-G-Code-Sender #1183: Error with files containing a naked G0

curt999, 2019-02-01:
## Problem description 
When I try to send a file containing a naked G0 I get the following error:  "Error while starting file stream: null"
If the G0 is removed the file sends fine.

### Steps to Reproduce the Problem
Here's a short file which causes the error:
%
(TOOL/MILL,0.1,0.05,0.000,0)
(FILENAME: )
()
G21
G90
G0X0.000Y0.000Z10.000
(TOOL/MILL,6.3500,0,1.0000,0.0)
M6 T201
M3 S12500
G0X-10.000Y10.000
G0Z10.000
G1Z-1.000F635.0
G1X10.000F1905.0
Y-10.000
X-10.000
Y10.000
G0Z10.000
G0Z10.000
G0X-15.000Y15.000
G0
G1Z-1.000F635.0
G1X15.000F1905.0
Y-15.000
X-15.000
Y15.000
G0Z10.000
G0Z10.000
G0Z10.000
M5
M30
(END)

If the naked G0 is removed (line 21) then the file loads fine.

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Jul 23, 2017
Product Version: NetBeans Platform 8.1 (Build 201510222201)
Java: 1.8.0_202; Java HotSpot(TM) 64-Bit Server VM 25.202-b08
Runtime: Java(TM) SE Runtime Environment 1.8.0_202-b08
System: Windows 7 version 6.1 running on amd64; Cp1252; en_US (serverlaunch)

#### Platform
  GRBL

whitetd, 2019-02-01:
G0 still needs info on where to go.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

AbySet, 2019-02-01:
**Whitetd**, What are you doing with those links ?

**Curt999** Here is full info how to use G0 whitch should not be nacked:

G0 Rapid Move
**Format** G0XxYyZz (This mean G0 should not be nacked).
X,Y,Z = axes on a three-axis machine
x, y, z = distance in incremental units, or absolute coordinates, to which the axes specified by the 
 receding upper case letter is designated to move.

**Purpose** Moves one or more of the axes, at the rapid speed, to a specified location or a
specified distance away from the current position. Location is designated by absolute coordinates when this command is preceded by the Absolute Coordinate command. 

Distance is specified by incremental units when this command is preceded by the Incremental Coordinate command.

A rapid move consists of a move of the X and/or Y axis along the XY plane and a move of the Z axis up or down.

If the Z-axis move is Directed from a lower point to a higher point, it will be executed before the XY move; if it is directed from a higher point to a lower point, it will be executed after the XY move.

Rapid speed is indicated in the .NC file by the Feed Speed command. F commands are ignored by the interface program when Override Program Speed is selected. When this selection is made, the rapid speed specified in Set-Up will be used instead.
Example 
G90
G0X2.00Y2.00Z2.00
The first command in this set indicates absolute motion, so the rapid move that follows (line 2) is made to the absolute coordinates (2,2,2). The axes move to a position that is two units away on each axis from the software home.

AbySet, 2019-02-02:
You'r welcome.
hmmm. then maybe it's supported on different cnc processor but not on GRBL.
Please let us know what they say about it, it may be usefull to know.

curt999, 2019-02-02:
Carbide 3D issued a ticket for this and said they would fix in a future version.

AbySet, 2019-02-02:
Ah ok, it's their issue then.
Thank you so much.

closed 2019-02-07 by breiler