winder/Universal-G-Code-Sender #1184: Tool probe module

Psychokiller1888, 2019-02-02:
## Problem description 
Probe plugin retract amount is wrong. Either the title is wrong and should say "Retract position" or the engin is wrong and should increment Z by the value and not travel to Retract amount.

https://puu.sh/CG9NQ.png

### Expected Behavior
After probing, the tool should travel on Z+ by the "Retract amount" value

### Actual Behavior
At the moment, after probing, the tool travels Z+ to said position, if its higher than actual

### Steps to Reproduce the Problem
1. I have a 14mm high tool Z probe
2. In Probe module settings, try to put retract **amount** 2
3. Under Z, set the touch plate thickness to 14 and probe direction let's say -5
4. Initiate the probe

5. Tool probing works, but the machine will stay on the touch plate instead of traveling up by the given amount

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Dec 18, 2018

#### Operating system
Windows 10

#### Platform
GRBL 1.1f

#### Other
https://puu.sh/CG9NQ.png

AbySet, 2019-02-02:
First i have to say i'm just a user like you trying to help people.
And yes you are right, they should change a name but looks like they choosed [amount] meaning from token zero position not from top of touch plate.

Psychokiller1888, 2019-02-02:
Which is mechanically not correct, as the probe module is like an Heidenhain probe cycle in which case values refer to the tool

AbySet, 2019-02-02:
It depends on the owner now.

tkms002, 2019-05-27:
how do I get this PROBE MODULE??  My UGS 2.0 only has the PROBE HELPER plugin.

breiler, 2023-08-16:
The retract amount is how far to retract after the first fast probe cycle before running the slow probe cycle.

I am going to add a tooltip with a explanation. If you have any suggestion on a better label for this please let me know.

closed 2024-02-07 by breiler