winder/Universal-G-Code-Sender #1185: Jog controller step size

Psychokiller1888, 2019-02-02:
## Problem description 
When setting a step size, if you don't select another text box, the old value will be executed instead of actually input value

### Expected Behavior
Refresh to actual value to avoid z crashes

### Actual Behavior
Could result in table crash. Did already the first time, when I typed 1 in z travel after fast approchaing with 5. Forgot to change selection box and jogged Z-. Tool ended 4 millimeter in the table

### Steps to Reproduce the Problem
1. Open jog controller
2. Move Z down without changing jog value. Keep the step size Z in mind though
3. Change the step size value to 1 by clicking in the box, not using the arrows, using your keyboard. Do not click in another box after typing your new step size
4. Click on Z- to jog down (mak sure you have space enough!!
5. The machine will travel by the last value and not by 1!

### Specifications
#### Version  
  UGS Platform 2.0 - Nightly Build Dec 18, 2018

#### Operating system
  Windows 10

#### Platform
  GRBL 1.1f

AbySet, 2019-02-02:
Programmers need to put a code that read the entered value somewhere in a fields called events, when you write something into the text box, the computer don't know if you finished to write or not, it need something to happend (an event) to read the value, i am actualy surprised to know that the value is not token after Z jog click event.
I suggest to programmer to put the reading code in text box lost focus event to fix it.
Until then, just choose another text box.

Psychokiller1888, 2019-02-02:
That's not true, onChanged events exit....

jahnj0584, 2019-02-03:
are you sure you have the latest version? This was a bug in 2017 that I had pointed out and was fixed.

Psychokiller1888, 2019-02-04:
Yes, I'm pretty sure, as long as the title returns the correct version I used the webiste provided link to download the latest beta

jahnj0584, 2019-02-05:
yea, its in the new version. bummer that it came back. its dangerous. Im using april version without that bug

Psychokiller1888, 2019-02-06:
It's..... very dangerous, crashed a few times at first, now I really make sure to click in another text field. And I do confirm that clicking the jog button does not refresh the value. I've taken a look at the code, it's as simple as a listener wrongly placed. I'll shoot a PR if I get the time to do it

breiler, 2019-02-07:
Sorry, github auto closed the issue because I typed "Fixes #1185" in the comments.

As I understand it, the nightly builds aren't up and running so we need to wait for this to be fixed before we can verify this.

Psychokiller1888, 2019-02-07:
No problem! Thanks for the quick fix!

Psychokiller1888, 2019-02-08:
I've been checking. Travis does compile ok, but your travis.yml does not specify to upload the builds...

https://docs.travis-ci.com/user/deployment/releases/

closed 2019-04-24 by breiler

Psychokiller1888, 2019-04-24:
Nightly builds fixed?

breiler, 2019-04-25:
Nope, not yet. There's a new manual build.

commit 80880302 (2019-02-07): Merge pull request #1188 from breiler/bugfix/1185-spinner

Fixes #1185 - make sure the spinner component issues change events
commit c28210bb (2019-02-07): Fixes #1185 - make sure the spinner component issues change events every time the text field is modified.