winder/Universal-G-Code-Sender #1186: Gcode edit

Psychokiller1888, 2019-02-05:
## Problem description 
After using Gcode edit, the edited code, even after saving (closing the edit tab) is not taking into consideration

### Expected Behavior
The gcode to be reloaded

### Actual Behavior
Old gcode is ran, resulting in badness.....

### Steps to Reproduce the Problem
- Open a gcode file
- File->Edit gcode
- Change some lines. I changed my pcb drill pointing depth that was wrong, from z-1.7 to z-0.3
- Close the tab, accept the popup asking for saving
- Run gcode

### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build Dec 18, 2018

#### Operating system
Windows 10

#### Platform
GRBL 1.1f

khturnings, 2019-02-13:
This would be a good idea

revtmyers, 2019-08-25:
Need to reload the Gcode manually.

closed 2019-10-08 by breiler