winder/Universal-G-Code-Sender #1187: Internal web browser

Psychokiller1888, 2019-02-05:
## Problem description 
Internal browser is... deprecated and does not render anything css related

### Expected Behavior
The pages to load correctly

### Actual Behavior
Loads a non stylized window: https://puu.sh/CHIy6.png

### Steps to Reproduce the Problem
- Help->About
- Click on the link to report issue
- Admire the splendid blue page as on my screenshot

Consider using a call to local browser, easy in java

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Dec 18, 2018

#### Operating system
Windows 10

#### Platform
GRBL 1.1f

#### Other
https://puu.sh/CHIy6.png

breiler, 2019-02-21:
This is a NetBeans specific issue, it uses a simple internal web browser and what I could see it can't be changed. I'm going to close this, but I hope this will be updated now that Apache will incubate the Netbeans project.

closed 2019-02-21 by breiler

Psychokiller1888, 2019-02-21:
What about not using the internal browser then and simply call browser.open() so it starts the system default browser?

breiler, 2019-02-21:
The about dialog is built in to NetBeans, we are not able to change that code.