winder/Universal-G-Code-Sender #1190: Not Detecting Serial Port

WaltRh, 2019-02-10:
## Feature request

U
## Problem description 
using a Dell laptop, with Platform UGS, while connecting to the Bobs CNC E3 there are no options for the serial port even when refreshed. Using a USB connection. Tried all 3 on laptop.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

WaltRh, 2019-02-10:
Please any help would be appreciated, looking forward to working with this machine, but stuck at the moment.

JacksCrafts, 2019-02-10:
Did you hit the refresh button next to where it says port?
Look in your device manager to confirm your Comport. .. COM 3 or COM 4

WaltRh, 2019-02-10:
Yes, I hit the refresh each time, but nothing shows up...strange

JacksCrafts, 2019-02-10:
What are you on Windows 10, using Arduino with CNC shield?
If you right click the start button you should see a short cut to Device Manager.
In there you can see if you com port is connected.

WaltRh, 2019-02-10:
I am using Windows 7 and Platform UGS

JacksCrafts, 2019-02-10:
You can probably type in Device Manager down in the lower left corner.
And is your buad set at 115200?

JacksCrafts, 2019-02-10:
If you go to your control panel.
Then hit hardware you will see Device manager.

JacksCrafts, 2019-02-10:
[image]

JacksCrafts, 2019-02-10:
If you don't have Ports (COM) in your device manager you may need to flash GRBL onto you arduino.

WaltRh, 2019-02-10:
JackCrafts,
Thank you for your help. I didn't have Arduino installed, I then installed
it and found the port it was in and typed it in, which allowed me to
connect and home the machine. So far so good, now I just have to learn to
use it, so I may have more questions coming.
Thanks again!!

JacksCrafts, 2019-02-11:
You are welcome,
Glad I could help.

breiler, 2019-02-19:
Are you satisfied with the work around?

Otherwise you could try to change the connection driver. This should allow the program to automatically detect the port.

See instructions here: https://github.com/winder/Universal-G-Code-Sender/issues/940#issuecomment-392761818

misan, 2019-02-21:
Same problem here with Mojave OSX. Arduino sees the ports, CoolTerm sees the ports but not UGS nor ugsplatform nightly builds see any serial port in my computer.

breiler, 2019-02-21:
> Same problem here with Mojave OSX. Arduino sees the ports, CoolTerm sees the ports but not UGS nor ugsplatform nightly builds see any serial port in my computer.

Have your tried switching to the  jSerialComm driver (#940)?

misan, 2019-02-21:
ugsplatform using JSerialComm works as suggested in #940

breiler, 2019-03-02:
Closing this as it's a duplicate of #940

closed 2019-03-02 by breiler