winder/Universal-G-Code-Sender #1192: New pendant

breiler, 2019-02-12:
I'm working on a new web pendant implementation and would like some feedback on my work this far.
Is there any missing must-have features?

### Waiting for connection
When loosing connection with the backend server this dialog will pop up:
[image]

###  Connection
If we aren't connected to a controller this dialog will allow us to connect:
[image]

### Main window
Main window for controlling the machine:
[image]

[image]

### Open and sending
Opening a new file and start sending:
[image]

[image]

### Jog controller
Jog controller, how should the step sizes be controlled?:
[image]

[image]

### Macros
[image]

[image]

winder, 2019-02-13:
Wow this is really nice!

khturnings, 2019-02-13:
one minor item I would like is in the 'Send File' could it also tell you the line number and last gcode line it sent

aleslukek, 2019-02-13:
looks great, great new features! Can't wait to try it out!
Will it work with get-requests as well? old custom gcode commands will still work, right? ( ".../sendGcode/?gCode=M4%20S1" for example)

Current pendant has some issues with "sent rows" number. It does not match sent rows to UGS platform number. Especially if there are comments or/and empty lines in gcode file. So calculating progress that is based on row numbers returns wrong number if there are comments in file.

Would there be an option for pendant autostart on UGS startup?

I built a physical control board that works through this pendant. It uses get commands to control it. There are certain features missing (soft reset for once... - i couldn't get it working, have some troubles with maven) but this would be awesome, could improve my board.

breiler, 2019-02-14:
@khturnings you mean like in the desktop versions where we display Sent rows, Remaining rows. I had it there to start with, but there is not any good way to determine which state we are in. So sending a file and jogging are using the same state which will make the line status report a bit wonky, so I didn't add it... I can have a look at it again.

If you mean to display the actual Gcode being sent would be implemented as a hack at the moment as we are polling UGS for information. For that kind of feature I think it would be better to wait until we have web sockets in place that could handle event driven communication.

@aleslukek I felt that it was best to split the different functions in to their own controllers, so unfortunatley the contract has changed a bit. I'll make sure to have a RPC-function for sending custom gcode. I'm using a pattern where fetching data is done with GET-methods and sending/updating data is done with POST. Will this be a huge deal for you?

I actually worked on an autostart setting last night, so that will be available as well. =)

There will be a specific action for soft reset that you can use, but initiating probe wasn't on my list as it's not available in the "core"-functions. Couldn't this be added as a macro instead?

Thanks for the feed back!

aleslukek, 2019-02-14:
I will work around it. Will change the code for different get requests and parse POST responses, no big deal. 

If there is a way to send gcode, most of the issues goes away. Sending multiple gcode lines is no problem. So probing could be initiated this way. UGS commands (send file, soft reset, connect, disconnect,...) however are, since if they are not preprogrammed there is no way of using them.

Two more questions, if UGS has an occasional error popup (not grbl error, but those pesky "error while processing response" - that are not as common on windows and osx but on raspberry pi), is this in any way communicated in this pendant? I am aware that those errors might be impossible for plugin to read, and could be handled a bit better in UGS (especially if they stop a job without putting grbl into a sleep mode for safety reasons). Just asking:)

Second question: sometimes UGS stopped responding to multiple (usually in quick succession) gcode lines sent via pendant. On console it showed it sent the code, but grbl did not sent a conformational "yes" and machine did not act on any of those codes. I only observed this issue while using pendant and not native UGS. This could also be UGS, console or grbl issue so I just wonder if anyone else observed the similar problem. Soft reset usually helps.

Thank you for your answers, greatly appreciated.

khturnings, 2019-02-14:
@breiler Thanks for the reply - I see what you mean think it would be best to wait for web sockets 👍

breiler, 2019-02-14:
@aleslukek I don't show any error messages yet. But my plan is to add web sockets for these kinds of features. And that is actually another problem I need to resolve, I think we need to change the behaviour of error messages popping up and instead make them visible from the console.

I've never experienced UGS to stop responding on multiple gcode lines. I'll give it some tests to see if I can reproduce the problem.

karoria, 2019-04-07:
Great pendant! For jog steps, you may put 5 buttons in a row below the jog buttons. Thosemay be for values in the multiples of 10 which is universally accepted for industrial cnc milling machines. Say,
0.001
0.01
0.1
1
10
100
I am talking this for mm. You may define them for inches differently.

breiler, 2019-04-11:
The new pendant is merged and will be available in the next build.

closed 2019-04-11 by breiler

J-Cat, 2020-12-17:
I'm trying to figure out how to configure Pendant to autostart?  Is this available yet?

Thanks!

breiler, 2020-12-17:
@J-Cat go to preferences and "Sender Options" then "Autostart Pendant on startup":
[image]