winder/Universal-G-Code-Sender #1193: UGS locks up and continues jogging (video link in description)

407pilot, 2019-02-15:
## Problem description 
UGS locks up and continues jogging when using touchscreen.

### Steps to Reproduce the Problem
Tap jog in any direction 2+ times using touchscreen. Machine will initially obey the jog command, then will take off using smaller jogging steps. 

See video link at bottom to see machine and UGS behavior. Frankly, I don't really know how to describe what it's doing. 

#### Version
UGS Platform 2.0 - Nightly Build Dec 18, 2018

#### Operating system
Raspbian Stretch on Raspberry Pi 3b+

#### Platform
Inventables X Carve
X Controller
GRBL 1.1

#### Other
https://youtu.be/d0cewo06HfY

407pilot, 2019-02-15:
I shutdown and power cycled everything and now it won't reproduce. While this is definitely related to the touchscreen, I don't think it's solely a touchscreen problem since the jog distance changed and the program became unresponsive, even to regular mouse clicks. I'll keep playing around to see if I can get it to reproduce.

breiler, 2019-02-15:
This could have something to do with the continous jogging feature. If you find anything please let us know!

AbySet, 2019-02-16:
On windows when we click and hold at any direction, it does those small steps, that's normal behaviour, its a feature in the program, see your OS touch screen config, maybe it's set to switch to hold when you double tap, i personaly don't know how your OS work, but you can look for eventual config that make double tap do a holding tap.

pmckeon, 2019-05-26:
I've had this happen to me a few times on Windows when repeatedly clicking a jog button in any direction, maybe it sometimes fails to detect the mouse up/button up event and will jog indefinitely until stopped.

AbySet, 2019-05-26:
I'm just a user like you @pmckeon, allow me to recomand you to make a new post about your issue, you are using windows (In this post @407pilot is using a Raspbian Stretch on Raspberry Pi 3b+) the problem may have different reason and i believe you will have better help with a new post.

closed 2019-10-26 by breiler