winder/Universal-G-Code-Sender #1195: No nightly builds since Dec 18, 2018?

mike-pittelko, 2019-02-22:
There haven't been any nightly builds published since Dec 18, 2018, and there have been code changes since then, when will the nightlies start up again?

rmisko11, 2019-02-22:
I am wondering the same thing?

timon2jkl, 2019-02-28:
Do we now have to try to compile our own? I would like to try a version whiteout the multiple errors on the probe

JimF42, 2019-03-17:
Still no nightly prepackaged builds... Am I looking in the wrong spot?

mike-pittelko, 2019-03-17:
You're looking in the right spot.  One of the admins needs to do it with their OAUTH token.  I think they need to add this:
----
deploy:
  provider: releases
  api_key: "GITHUB OAUTH TOKEN"
  file: "FILE TO UPLOAD"
  skip_cleanup: true
  on:
    tags: true
--------

To the travis.yml file, I'd put together a PR for it, but it's kind of useless to do so without their api_key.... that I don't have.

I don't think they realize how difficult it is to build this thing if you don't normally do java development and have all the dependencies and build tools available to do so.  I think the only way around it is to either do that, or create a fork and do the right thing in github+travis on your own.  Wish they would at least mark the "nightly" as out of date.

mike-pittelko, 2019-03-17:
Will get a pr for this....but in the meantime

Add to .travis.yml
deploy:
  provider: pages
  skip-cleanup: true
  github-token: $GITHUBAPIKEY  # Set in the settings page of your repository, as a secure variable
  keep-history: true
  local-dir: ugs-platform/application/target
  on:
    branch: master

Then, in the travis settings for the repo, set the variable "GITHUBAPIKEY" to AN apikey set up on the github repo.

mike-pittelko, 2019-06-29:
Would be great if they would at least mark the "nightlies" as not updated since last year.....

JacksCrafts, 2019-06-29:
Product Version: Universal Gcode Platform 20190423 
April this year is the platform I am using.  It has a lot of improvements.

winder, 2019-08-20:
This is finally resolved, the links in the README and on https://github.com/winder/ugs_website will now have the latest code.

closed 2019-08-20 by winder

breiler, 2019-08-20:
Wohoo! Thanks Will!

khturnings, 2019-08-20:
This is great - thanks for all your hard work