winder/Universal-G-Code-Sender #1200: head crash with platform

loonster, 2019-03-01:
I have been using Shapeoko XL for about 2 years without problems.  Emboldened by the prospect of a newer and "better" version of UGS, I downloaded the UGS Platform.  It had so many bugs and problems when I tried it on both Mac Mojave  and Ubuntu 18.04 Linux machines that I gave up and deinstalled it.  

Then a problem came up that I've never had before:  when I tried homing the CNC machine with UGS classic v2, the spindle moved rapidly down and crashed into the machine bed!   After finding no help online, I used Carbide Motion's  (v. 4) built-in utility to reset the GRBL 1.1f  and that solved the head crashing problem.  However, a new problem then surfaced:  I used  UGS classic to issue a z-probe command (G20 G38.2 Z-0.6 F2; G92 Z.062; G0 Z0.7;) and the carriage moved UP instead of down.   

Again, no help anywhere on any forums.  A day later, I again issued the z-probe command it now works!

JacksCrafts, 2019-03-02:
I have the same problems. The Z going in the wrong direction may have something to do with being too large of a Negative number. I just make sure to always zero out the Z before hit the macro and it works ok.  Once in awhile my machine (DIY) will jump 8-12 inches when I turn it on, don't know the cause. I also use arduino uno with grbl 1.1f.

loonster, 2019-03-02:
Heh heh.  I came to the same conclusion.

breiler, 2019-03-02:
@JacksCrafts it sounds dangerous with that much movement when starting. But it doesn't sound like it's the issue they describe over at GRBL: https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#connecting-grbl
Do you have any start up blocks configured (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#n---view-startup-blocks)?

@loonster sorry to hear about your "crash", but this doesn't sound like it's an UGS issue and I would like to close this. However I'm interested in any feedback you may have on the UGS Platform. Did you try the nightly build or the "stable"?

loonster, 2019-03-02:
I was using the nightly build of the UGS Platform.  However, since using it appeared to mess with how the Classic ran, I’m loathe to try it again.  How this might occur, particularly on the Mac Mojave, is unclear (note:  visualizer STILL doesn’t work on that machine despite working for 2+ years before the big snafu a couple days ago, although it functions fine on my Ubuntu 18.04)., 

BTW, I did read the two links you sent and found them to be exceptionally readable.  The vast majority of technical explanations on the internet are abysmal and even misleading.

JacksCrafts, 2019-03-02:
Product Version: Universal Gcode Platform 2.0 [nightly] / Dec 18, 2018
Java: 1.8.0_201; Java HotSpot(TM) 64-Bit Server VM 25.201-b09
Runtime: Java(TM) SE Runtime Environment 1.8.0_201-b09
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)
User directory: C:\Users\jacks\AppData\Roaming\.ugsplatform\2.0-SNAPSHOT\dev
Cache directory: C:\Users\jacks\AppData\Roaming\.ugsplatform\2.0-SNAPSHOT\dev\var\cache

JacksCrafts, 2019-03-02:
>>> $N
$N0=
$N1=
ok

JacksCrafts, 2019-03-02:
Now the other link could be the answer.
I have a cheap knock off Arduino Uno so the quality there may not be great.

And I know some folks do not trust the use of USB cables.
But I have been using UGS with Arduino/CNC shield a couple of years now and rarely have any trouble.
Thanks for the tips.
I♥UGS

msearle58, 2020-06-11:
I'm new to CNC so therefore new to UGS.  However I've been using the latest nightly build (20200425) for several weeks now and have also experienced Z axis crashes.   However I seem to have solved the problem by always going through these steps before I run ANY GCode...

1. Reset Zero.
2. Hard reset the Arduino by unplugging the USB cable.
3. Physically reconnect the USB and then reconnect in UGS.
4. Zero again.

I noticed the offsets against XY and Z have small numbers below which I assume are the XYZ co-ords reported by GRBL.   I seem to get the Z axis crashes when these numbers are different, typically when stopping a GCode run or doing anything that separates these values.   I therefore assume that UGS is not re-synching with the numbers within GRBL when a zero is sent.

I'm using a MAC running High Sierra, with and Elegoo Arduino running GRBL 1.1

BTW I think UGS is an excellent program it is so easy to use and the visualiser is brilliant!

mtembi, 2021-03-17:
I am new to the CNC world and use the UGS platform v2.0.7/Dec 02 2020 on GRBL 1.1f. 

I've had a problem with Z-axis crashing during probing. I realized when the WCS is G54 and probe WCS is set to G55, Z will always crash into the bed. Broke a couple of bits :(

I now always check UGS state WCS and probe WCS before probe.