winder/Universal-G-Code-Sender #1202: UGS ignores the feed in the code

rsilva87, 2019-03-13:
Not sure how to upload pictures but I write my code using mastercam 9.1 then UGS to send the file to the arduino uno flashed with grbl 1.1 to the 3 steppers I have on my mill. For some reason though, the feed always shows 0 even though my code very clearly states:

"G1 Z-.1 F8.21"

all the rapids are working correctly but when it actually goes to cut, the feeds just read "0"

## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

breiler, 2019-03-13:
Which version and variant of UGS are you using?

rsilva87, 2019-03-13:
2.0 (nightly build) ugsplaformx64 from netbean

breiler, 2019-03-13:
Which nightly build? The latest from 2018 18 dec?

rsilva87, 2019-03-13:
yes sir that is correct'

breiler, 2019-03-13:
I was able to reproduce your issue. Please check if your gcode is created with a G21-command (units in millimeters).  In that case your feed rate is too low to be shown in the DRO as it doesn't use decimals.

Using feed rate in millimeters is shown as "0".
`
G90 G21 G17
G1 x-200 F8.21
`

Using feed rate in inches is shown as feed rate "8"
`
G90 G20 G17
G1 x-200 F8.21
`

rsilva87, 2019-03-13:
that got her going again. thank you sir, that was driving me nuts. but now
i know!

closed 2019-03-13 by breiler