winder/Universal-G-Code-Sender #1207: Automate CNC with barcode

kikirikiki, 2019-03-27:
## Feature request

Hi, I would like to automate the process of my CNC. 
I would like to start a g-code from a Raspberri Pi with a barcode or QR-Code. Is there any way to do it? Any idea how I could do it? is it possible?

Thank you very much!!

breiler, 2019-03-27:
Do you have any programming experience?

In that case there is a web based API available that you could hook into from an external program. It's currently a bit limited. Have a look here: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/pendantui/PendantUI.java

I'm currently working on a new version of an API in #1203. If that gets merged you could pretty much automate a complete workflow using the API. Shortly after that I have a new feature where you can run UGS headless in a terminal which will probably be useful when running on a RPi (no popups, smaller memory footprint, etc.)

If you don't have any programming experience, I would say no - it's not possible. =)

kikirikiki, 2019-04-01:
And a simple way to do it? I want to automate 10 diferent G-Codes. I had the idea with a barcode, but another way would it work for it. Is there anyway to automatic start a G-Code between 10 diferent G-Codes? Would it work with a PLC? Any idea?

breiler, 2019-04-01:
You can run UGS from a terminal, it can take a filename as a parameter:
https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-cli

kikirikiki, 2019-04-01:
so, I could use a Raspberry Pi as a terminal with this programm, right?

breiler, 2019-04-03:
Yes in theory it should work.

closed 2019-05-16 by breiler