winder/Universal-G-Code-Sender #1208: $H: Alarm: Homing fail w/ An unexpected command was completed by the controller w/ Java Exception

toffuser, 2019-03-29:
## Problem description 
When sending $H to GRBL using UGS Platoform 2.0 nightly (Arduino UNOS with CNC shield) I get the error message Alarm: Homing fail after it moved for a while (Any axis, X, Y or Z)

### Expected Behavior
It is supposed to reach the limit switch for any axis.

### Actual Behavior
It moves for a while then after 10 seconds, it stops.

### Steps to Reproduce the Problem
  1. Open the program
  2. Click on Home Machine
  3. let the machine try to go HOME and FAIL

### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build Mar 29, 2019
  UGS Classic 1.0.9

#### Operating system
  Windows 10

#### Platform
  GRBL 1.1f

#### Other
The exception thrown in the UGS platform : 

com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:829)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:170)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:850)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:222)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:244)
	at java.lang.Thread.run(Thread.java:748)

[image]

breiler, 2019-04-01:
Could you post the settings for GRBL?

AbySet, 2019-04-13:
Hey,
Sorry i'm a bit late but i have been through a same issue and i fixed it:
- Your problem is not a UGS one, it's actualy a hardware issue caused by IME troubles and it needs a noise filter.
- To solve this you should add a filter to your limit switchs wiring by adding a 4.7K resistor and 100nF capacitor and here is how to do it :

[image]

Notice that the 3 resistors color code shows a value of 1K but you have to use 4.7K as labeled (I did not do the schematic).

Hope this help.

carneeki, 2019-04-13:
Adding to Aby's great answer, the 4K70 resistor colours are

4 band:
Yellow (4)
Violet (7)
Red (x100)
(any colour for tolerance, but gold is pretty common = 5%, brown = 1% and
silver = 10%)

5 band:
Yellow (4)
Violet (7)
Black (0)
Brown (x10)
(any colour for tolerance, but gold is pretty common = 5%, brown = 1% and
silver = 10%)

[image: image.png]
Image courtesy of https://www.electronics-tutorials.ws/resistor/res_2.html

Keeping the pull up resistors and capacitors physically close to the
Arduino inputs and with short wires is helpful to minimise interference -
we can think of wire acting as antennas picking up cruft, and the
capacitors are low pass filters that only let "slow changing signals" like
buttons or switch presses into the microcontroller.

toffuser, 2019-04-13:
Abyset is right!

I also shielded my limit switch cables. Make sure to put them far away from the cables used for the motors.

montakhab, 2019-07-23:
hi toffuser
did your problem resolve? i have same problem . I have used shielded cables for both limit switches and motors. I also have used normally closed end switches with with noise filtering as shown in github wiring limit switches and close to Arduino. Who has experience in this case to help me? Thanks in advance
[image]
[image]

montakhab, 2019-07-23:
and my GRBL settings. 
[image]

breiler, 2020-01-07:
@toffuser, was this resolved?

closed 2020-03-09 by breiler