winder/Universal-G-Code-Sender #1209: UGS Platform Shutting off half way through loading process

Tucksroom, 2019-03-31:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

I am having this issue in 2 different computers at this time. What is happening on both of them is I click the short cut to open UGS Platform(Nightly Build 2.0) It starts to load as usual then just stops and turns off. 
I am Using Windows 10 on a Lenovo laptop.
We are using GRBL 1.1
Our Machine is a Delusional Designs Freedom Series
 
Things I have tried to do to fix this issue...
Delete and Reinstall UGSPlatform
Delete the Config Settings and restart program
Delete the Config settings and reload the DDCNC Settings
Updated Java
Delete and reload Java
Update Video Card driver
Delete and reinstall Video card driver
Tried opening from the Command Line I get a console error. I do not know what it means but it suggests using a new console. 
Updated Windows
Updated Bios

I have run into some small issues from time to time with UGSplatform but most issues are fixed by updating Java or updating a driver. I am at a loss. 

Jim Tucker
DDCNC

breiler, 2019-03-31:
What version of Java are you running?

Tucksroom, 2019-03-31:
I am using Java 1.8 the windows offline version.

breiler, 2019-04-01:
Can you post the contents of the log file. 

Have a look here to get the location of "messages.log":
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

Tucksroom, 2019-04-01:
Is this what you are looking for?

 

Jim

 

http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

breiler, 2019-04-03:
Did you attach a file? It wasn't added by github.

Tucksroom, 2019-04-03:
[grbl.zip](https://github.com/winder/Universal-G-Code-Sender/files/3039206/grbl.zip)

Idk what happened but here is a zip with them all in it.

AbySet, 2019-04-13:
UGS 2.0 (May.2018 version) had this issue and did get fixed later, did you reinstalled the same version you had ? download the last version and try.

Also you may do this for cleaning up all settings:
- Set your windows folders settings so it shows hidden files.
- Open this folder: C:\users<Your name user>\AppData\Roaming
- Delete a folder with name: .ugsplatform
This will make you delete all cache memory of the program and clean it all (You will need to redo all your settings).

breiler, 2019-10-27:
Closing as there was no further responses.

closed 2019-10-27 by breiler