winder/Universal-G-Code-Sender #1211: Feature request: Support (MSG, ...) comments

ValdemarOrn, 2019-04-09:
## Feature request

It would be great if UGS supported a way of providing text feedback to the operator in the form of text dialogs or messages.

LinuxCNC implements this simply by handling any comment starting with (msg, ...) as a special type of message, and shows the remainder of the comment text as a dialog to the user. This allows the operator to insert comments at specific points in the GCode that will be displayed during the run.

It would be great if UGS could do the same. This would improve our ability to write tool-changing macros, to provide operator instructions after an M0 command, or anything else that requires manual intervention.

A description of LinuxCNCs behaviour can be found here: http://linuxcnc.org/docs/html/gcode/overview.html#gcode:messages

Last but not least, thanks to all the contributors for providing a great piece of software, your efforts are highly appreciated!

AbySet, 2019-04-13:
I agree, it could be good to add.
Some GCode files are compatible with UGS except for changing tool code T6, it would be nice to add the abylity to skip it or pause with a message (Skip would be used for testing the machine moves without physical carving/cutting).

brooksben11, 2023-01-23:
Came across this feature request when Googling whether such a feature existed in UGS. It'd be a really fantastic addition; for instance if you have a bunch of tool changes it could just TELL you what they should be rather than having to look them up/remember them.