winder/Universal-G-Code-Sender #1213: Serial connection unstable on RaspberryPI

breiler, 2019-04-13:
The new default connection driver jSerialComm does not work on ARM-based processors like on the RaspberryPI or Asus TinkerBoard. The connection can be established to GRBL but fails intermittently. 

We are using an old version of jSerialComm which might be the cause. But upgrading the library can't be done due to this netbeans-plugin-module issue:
https://github.com/mojohaus/nbm-maven-plugin/issues/42

Tried building a snapshot version of the nbm-maven-plugin (nbm-maven-plugin-4.2-SNAPSHOT) which now lets me build with the latest jSerialComm library.

We need to wait until Apache Incubation process of netbeans is ready and has packaged a new version (looking forward to using your work @ebarboni): https://github.com/apache/incubator-netbeans-mavenutils/

In the meantime you can switch to JSSC-driver on RaspberryPI which works reliable: https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi

winder, 2019-04-20:
Looks like we should be able to upgrade netbeans platform soon! Netbeans has been accepted to Apache, I think they’re still working on a press release but it’s been announced on the public mailing list:
http://mail-archives.apache.org/mod_mbox/netbeans-dev/201904.mbox/%3CCACkjAxS-fCp9mbr435QaZacbzKWpgfGgMdt0tCknUK4HOow4Hg@mail.gmail.com%3E

One of the last steps was getting the netbeans domain, which must be complete since they’re already staging artifacts 
http://mail-archives.apache.org/mod_mbox/netbeans-dev/201904.mbox/%3C003a01d4f50b$0b7ec610$227c5230$@apache.org%3E

breiler, 2019-05-09:
It seems that the new netbeans libraries are available on maven central. It might be possible to upgrade to the latest platform. I would like to give this a try later this week.

mike-pittelko, 2019-10-03:
Any progress on this?

breiler, 2019-10-03:
I made a search a couple of weeks ago and couldn't find the new version for building NBM-modules. Made a quick search for it now and found the new version which seems to have the required fix: https://mvnrepository.com/artifact/org.apache.netbeans.utilities/nbm-maven-plugin/4.2

I'll give it a go and try to update it.

jes, 2020-11-11:
Possibly related to this: I have found spurious disconnections using JSerialComm which went away when I switched to JSSC, and I'm not using an ARM board, it's a normal x86_64 laptop (Thinkpad X200).

breiler, 2020-11-11:
> If the JSerialComm driver doesn't work properly on Linux, should we consider making JSSC the default?

@jes We have had far more problems with JSSC than JSerialComm (especially in Windows) so I rather not make it default again, you can search for issues containing "JSSC" in project.

The JSSC project looks abandoned, but there is a fork that seems to have active development which we could give a try: https://github.com/java-native/jssc

jes, 2020-11-11:
Fair enough. I wonder if it would be easy to have a separate default for Windows vs Linux?

It doesn't make much difference to me now as I now know what the problem is, but I can't be the only person who has spent days trying to debug an apparent EMI problem only to find out it is a broken serial library.

breiler, 2020-11-11:
Sure, we can try to add that. I have not had any problems on my Linux instance with JSerialComm so I hope we don't introduce other problems... =/

So the default library to use should be according to the following:
Linux - JSSC
Windows - JSerialComm
MacOSX - JSerialComm

jes, 2020-11-11:
In case it helps, here's a simple test case that I find reliably fails with JSerialComm and passes with JSSC:

 - reduce Grbl's baud rate to 9600 (BAUD_RATE in `grbl/config.h`)
 - make a G-code file that just moves to (0,0), (0,2), (2,2), (2,0), and back, over and over again
 - send the file using UGS. It will go offline within 5 seconds

You don't even need to be connected to a CNC machine, just running Grbl on an Arduino with no hardware connected.

breiler, 2020-11-11:
Excellent thanks!

breiler, 2020-12-05:
I'm closing this as it is the best we can do right now.

closed 2020-12-05 by breiler