winder/Universal-G-Code-Sender #1214: Build release package fails if GRBL device is not plugged in and ready.

thawkins, 2019-04-14:
## Problem description 
Running package build without plugged in and active grbl device results in test failure and build abort. 

### Expected Behavior
Package should be built

### Actual Behavior
Apr 14, 2019 1:38:50 PM com.willwinder.universalgcodesender.GrblController rawResponseHandler
SEVERE: Error while processing response <Grbl 1.1f>: java.lang.Exception: Cannot send command(s), comm port is not open.
java.lang.RuntimeException: java.lang.Exception: Cannot send command(s), comm port is not open.
	at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:239)
	at com.willwinder.universalgcodesender.GrblControllerTest.errorInCheckModeSending(GrblControllerTest.java:1429)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.apache.maven.surefire.junit4.JUnit4Provider.execute(JUnit4Provider.java:367)
	at org.apache.maven.surefire.junit4.JUnit4Provider.executeWithRerun(JUnit4Provider.java:274)
	at org.apache.maven.surefire.junit4.JUnit4Provider.executeTestSet(JUnit4Provider.java:238)
	at org.apache.maven.surefire.junit4.JUnit4Provider.invoke(JUnit4Provider.java:161)
	at org.apache.maven.surefire.booter.ForkedBooter.invokeProviderInSameClassLoader(ForkedBooter.java:290)
	at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:242)
	at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:121)
Caused by: java.lang.Exception: Cannot send command(s), comm port is not open.
	at com.willwinder.universalgcodesender.AbstractController.sendCommandImmediately(AbstractController.java:504)
	at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:236)
	... 29 more

### Steps to Reproduce the Problem
 mvn package assembly:assembly

### Specifications
#### Version
ugs_platform, git checkout on date of submission. 

#### Operating system
Fedora 29

#### Platform
GRBL 0.9 (Woodpecker CNC)

breiler, 2019-04-14:
Can you post the complete build log. This error message in the log is expected in the test and shouldn't stop the build.

thawkins, 2019-04-14:
I will do another run and capture it this evening.

carneeki, 2019-04-28:
I just came across the same error while running `mvn install`.

**OS**
```
$ uname -srvio
Linux 5.0.9-arch1-1-ARCH #1 SMP PREEMPT Sat Apr 20 15:00:46 UTC 2019 GNU/Linux
```
**Java**
```
$ java -version
openjdk version "1.8.0_212".
```

The build usually works, and fails only occasionally, even when running the exact same build command a second time. I have attached a typescript for analysis, though this is one where the `mvn install` command worked, though we can see on lines 1330, 1368 1433, and 1471 all show that tests are trying to connect to Grbl and it fails.

[typescript.txt](https://github.com/winder/Universal-G-Code-Sender/files/3124658/typescript.txt)

breiler, 2019-10-27:
Those messages are expected and is a part of the unit test where different parts are mocked to provoke different errors. In these cases that you've pointed out the mocks hasn't been properly set up but doesn't matter for those specific tests. I'll try tweak them so it isn't so misleading.

If the build fails on a certain test it's usually when we are testing something that's timing sensitive, such as testing threaded code. If that happens I'm interested in that log!

breiler, 2019-10-27:
I'm closing this as I can't reproduce it, please reopen if it still occurs.

closed 2019-10-27 by breiler