winder/Universal-G-Code-Sender #1215: UGS Freeze/Stops

stereolab54, 2019-04-18:
I have a recent issue of UGS freezing midway through CNC jobs. The issue is unpredictable (occurs 13 minutes or 90 minutes in a 2 hr job) despite having updated the Xms value from 64 to 256 as well as changing my Windows power settings to never shut off.  I'm using version 2.0 from a nightly build dated 6/3/18. I have also used UGS stable version 2.0 and have the same issue. Has anyone else experience this?

AbySet, 2019-04-19:
. If you are using a USB connexion it may have troubles or getting not stable, this happend when USB cable is bad or USB port connector (machine side or PC side) that have soldering lose or bad connector.
. If you are using Bleutooth connexion as serial port, you may avoid anything between the machine and the PC that can block the signal, long distance between them can also do it resulting on a week signal and lose.
. Some people are using chinese cheap power supply that can cause problems especialy if using a manual spindle speed control device with it. Using a desk PC power supply is much better choice.

To help you more, we need to know more, all software specs (OS and UGS versions), as well as hardware specs (type of CNC coltroler and its features).

breiler, 2019-04-23:
What processor architecture are your computer running? We've seen problems with ARM processors on Linux, could be similar problems on Windows.

stereolab54, 2019-04-26:
I am running Windows 10 (version 1809) on a Dell Inspiron 3147 laptop. Processor is an Intel Pentium 2.16GHz. Fresh reset/reinstall of Windows on 4/22/19 as previous version (1709) wasn't updating. UGS version is the nightly build from 4/23/19. I was unable to access Windows Device Manager prior to the reset so could not verify if my USB ports were powering down during the job.  On 4/24 I ran my job twice, to completion, with dust collection and router turned off.  About 69900 rows which took nearly 2 hours each time to finish.  Tonight, 4/25, I ran the job (with dust collector and router on) and it froze.  I checked my USB ports to verify they stay on and both are set to not to turn off to conserve power.  My laptop was plugged in the entire time (for all three runs) to a grounded power strip.  The fact I could run two (basically back to back )jobs with no problem and tonight suffer the freeze again is very odd.  I could blame the router and dust collector but the UGS software was unresponsive.  I put the job on pause but could not "play" it again.  Needed to shut down the program.  Are there tests I could run to help isolate the issue?  Any other pieces of info I can provide? Thanks.

breiler, 2019-04-26:
Could you check which connection driver you are using, try switching to the other

[image]

stereolab54, 2019-04-27:
I had it on JSerialComm and changed it to JSSC.  Tried it about 4/5 times last night (and again this morning, 4/27) without any change.  It still froze.  Other option for connection driver is TCP which did not connect the machine.  Tried both USB Ports, COMM3 and COMM4, no difference.

AbySet, 2019-04-28:
Ok i was reading and reading many times your posts, let's gather all informations you provided:

**Problem:**
   - The UGS freezes within 2 hours job (like random time).

**System specs:**
   - OS: Windows 10 (v 1809).
   - Connection to the machine: USB (JSerialComm driver selected even tested with JSSC).
   - UGS version: 2.0 nightly build  (4/23/19).
   - Router type: **Unknown**.
   - Router speed controle: **Unknown**.
   - Dust collector: Used.
   - Machine controler type (processor): **Unknown**.
   - Power supply type for the machine: **Unknown**.

**Tests you have done:**
   - Ran 2 jobs with dust collector and router were OFF [Result: Succes].
   - Ran jobs with dust collector and router ON [Result: Freezing UGS].

**Conclusion:**
   - The data from PC to the machine is being lost or corrupted when the router or/and dust collector are ON, this mean can be caused by:
      . Router or/and dust collector motors could have bad internal capacitors whitch are not filtering anymore and when they spin they create bigger sparks on the rotor part whitch will disturb electronics (*).
      . Wires from the controler to the steper motors are not well shielded, if they are not they will act like antena and absorbe interferences, the shield wire must be connected to the GND on the controler box but not on the machine body, connecting both sides can cause a loop circuit and get you to more troubles (*).
      . Some cheap power suplies have two separate GNDs by their design and are not recommanded to use.
      . I have seen people having troubles when using cheap router speed controler and cheap power supply in same time.

**Missing information:** 
   - The machine type and its processor (arduino based board or so).

(*) Most causing the issue.

AbySet, 2019-04-28:
You may take a look here: https://www.youtube.com/watch?v=UpJ4txXh3WM

stereolab54, 2019-05-02:
I'll review your suggestions as well as view the video (thanks for sending them).  What I have below may either strengthen your belief that it is wiring and not the program at issue (or not).

I have done additional tests and it appears I can get one good run before things start freezing. I am using another PC, newer, (Windows 10, 1809).  Over the weekend, I was able to CNC both my files without issue.  Thinking I may need to get a replacement machine, I tried mine again.  Here are the results:

- Original PC - Vacuum (dust collector) off, router off - OK
- Original PC - Vacuum on, router on, (moved the mouse, thinking the machine is sleeping) - Froze
- Original PC - Vacuum on, router on, (cleared cache, moved the mouse, thinking machine is sleeping) - Froze

- "Other" PC - **One referenced above** (which worked before) - Vacuum on, router on, (moved the mouse, thinking machine is sleeping - Froze
- "Other" PC - Vacuum off, router on, (cleared cache) - Froze
-"Other" PC - Vacuum off, router off - Froze - **this is strangest of all.**  

It seems the more I try to run UGS the more likely the machine/PC is to freeze, as if something in the background is building up errors until it no longer runs.  I'm no electronics person (which is why I will watch the video), however the program itself freezes, meaning the GUI doesn't allow me to take any action at all. 

As for my machine, using a BobsCNC E4.  Site references it uses Arduino controller, however "Arduino" is not written on the board.

winder, 2019-05-02:
Thanks @AbySet for the analysis!

@stereolab54 try increasing the memory allocated to UGS as described here:
http://winder.github.io/ugs_website/guide/troubleshooting/#program-slows-down-and-send-freezes

winder, 2019-05-02:
This unresponsiveness issue has been reported from time to time in the past, but I've never heard of it NOT happening during dry runs, that's a pretty big clue.

One more question that I have after re-reading your last post. When you say `Froze` do you always mean UGS has become unresponsive? In the EMI error scenario, the USB communication is interrupted, so the program stops sending and UGS can no longer control the CNC machine. Both of these issues could be considered a "freeze".

AbySet, 2019-05-02:
I have checked the manual of your machine and yes it is based on Arduino uno controler
[image]

You may check if that Arduino is an original one because people are reportings troubles with the clones ones, take a look at the picture below:
[image]

Also check if this part or square plastic is broken or missing:
[image]

AbySet, 2019-05-02:
You can see on the first picture that the steper motors wires are not shielded at all and the motors drivers does not have heatsink to cool them down, when running for a long time they get extremly hot and they have a shutdown circuit to cut them off when they reach the limit of max temperature, you can buy heatsinks separatly if you want.
The following picture shows a much better motor driver with its heatsink, the reference is DRV8825.
[image]

khturnings, 2019-05-02:
I have been following this with interest as I have been having simular issues - wierdly it freezes only on one axis.
I am not going to hijack this post just thought I would mention it and try the fixes you are suggesting.

Abyset - you mention unshielded wiring - I have got some to replace my wiring but I am unsure how to ground the shielding - any advice?

stereolab54, 2019-05-02:
Huge thank you for all the assistance.  @winder, I did increase the memory when I first started experiencing the issue with the "original" PC but have not, subsequently, done so since getting it back. I will do that though. Also, when I say "froze" I am referring first to the controller (no axis movement), but more recently to the interface.  Run time may continue to calculate, but if I click Pause or Stop, nothing happens in the console, "job paused" and the like.  I need to close UGS, but also restart the PC, as UGS connects back to machine (if I don't shutdown) but doesn't let me home or jog.   @AbySet, good call (the heatsink).  I had originally purchased the E3 machine and it ran without issue for many months.  I recently upgraded to the E4 and needed to use parts from the E3.  One of those was the controller, which had the four heatsinks attached to the motor drivers.  One fell off and since the photo instructions didn't include them (and I'm not too bright) I didn't reattach it (lost it's adhesive).  That might explain why I can get one good run (or two) but after that things stall?  Need to get some heat tolerant epoxy.  I'll keep everyone apprised of my progress.  Thank you again!

carneeki, 2019-05-02:
@khturnings - it depends a little on the physical connectors you have. If
you have the popular GX16 ("aviation") connectors, the shielding can be
folded back around the outside of the outer cladding (thick insulation),
the conductors (wires) soldered to the pins and then the shielding is
sandwiched between the connector and the cladding. The electrical path for
the shielding to ground is then through the connector into the chassis, and
from chassis to ground pin. This is similar to how coaxial cable for TV and
radio is shielded.

Another trick is to roll the shielding into a wire and solder that to a
dedicated pin in the connector (example: you have a 4 pin stepper, so 5 pin
connectors are used with the 5th pin going to ground). Personally, I have
not used this method though.

On the topic of grounding the shielding, and I'm assuming you're connecting
to an iron machine such as a lathe or mill, it may be worth connecting the
shielding to ground at only one end in order to prevent something called a
"ground loop". This is where stray interference can take multiple paths to
get to the electrical ground. Further reading (and one or two diagrams in
the Solutions section here might be of interest)
https://en.wikipedia.org/wiki/Ground_loop_(electricity) Because the
castings of machinery can be used as a ground, it is common to have limit
switches and probes connect to ground via the machine, which then has a
dedicated ground lead to the ground terminal on the controller. If the
stepper cable provides an additional path to ground at each end, this may
produce those ground loop issues, but at just a single end, the loop is
open.

stereolab54, 2019-05-10:
Used my machine (original PC) tonight after a week of not doing so.  Took @AbySet advice and reattached the heatsink.  First gcode file ran fine (vacuum on, router on), total time 1 hour and 51 minutes.  Second file froze (vacuum on, router on - app stopping working, though time to completion continued to run).  Closed UGS, cleared cache and restarted application.  Unable to connect, received error message "no homing device set for this method".  I hate to ask (knowing it is bad form), but is there another app I can use to determine if it is my machine or the app?  I have tried Openbuilds, however it closes on me referring to a bad port.  Perhaps bcnc?  Willing to continue to troubleshoot, but do need an alternative as this is getting frustrating.  Thanks for any advice offered.

AbySet, 2019-05-10:
Two major things can make huge difference:
- Shield your motors wires to the ground (Connected only to the controler ground).
- Add a fan to cool down the heatsinks since the working time is too big for small motors drivers.

To reconnect the machine make sure you have it sutdown and repowered, but i am surprised about the error message "no homing device set for this method", i have never seen it before.

But i found this in the APP itself:
**Program Slows Down and Send Freezes**
If you notice slowness while running your program, it may mean UGS is running out of available memory. There are a couple things to try:
1 - Check the controller settings, and make sure "Arc Expander" is not enabled. This can take a small program and turn it into a very large one by converting arcs into many small movements.
2 - Increase the memory allocated to UGS by navigating the the intsallation directory (See paths above). There is a folder named etc containing ugsplatform.conf, open this file with a simple text editor and modify the value of Xms to something like -J-Xms256m, or larger. _Additional details can be found here_ Link: 
http://wiki.netbeans.org/FaqSettingHeapSize

And this brings us back to @winder comment.

stereolab54, 2019-05-10:
I'll update the available memory again, but it didn't seem to matter when this issue first began. As for shielding, based on the image below, how would I go about doing that?

[image]

@khturnings detailed response (while clear to others) is leaving me confused. As for a fan, CPU or a cheap desktop model?  Apologies for my ignorance and thanks for your patience.

AbySet, 2019-05-10:
This video shows how the wires should be, this mean you need to change all motors wires whitch is hard to do i gess (in the video focus only on how wires are made, dont care about anything else).
https://www.youtube.com/watch?v=9Ztop0JCi94&t=386s

This video shows how to proceed when you dont wana change motors wires. But instead of soldering the green wire to the metal piece of the connector, you should plug it somehow to the negative of the arduino shield called GND (because you dont have that connector) 
https://www.youtube.com/watch?v=qYCiIURC7BU

For cleaner work you can buy one of these kind of conductive tube:
https://www.youtube.com/watch?v=WFi0VT1sTsU

So basicly what is a wires shielding ?
Make them covered by a metalic material that should be connected to the negative polarity of the controler, that negative polarity is called ground (GND).
The image bellow shows the pins you can use as negative GND:
[image]

stereolab54, 2019-05-29:
Wanted to comment and close.   Turns out it was my USB cable all along.  Replaced it using the one which came with the new controller I ordered and abra cadabra, I haven't had a issue with back to back carves.  Thanks to everyone for your patience and advice.  Take care.

closed 2019-05-29 by stereolab54

scratchboom, 2020-11-26:
The switch of the driver in UGS setting from JSSC to JSerialComm helped me

chrisbellcreations, 2021-03-01:
Just found this thread last night and tried putting a little fan on my controller and it seemed to fix the issue.