winder/Universal-G-Code-Sender #1228: UGS platform won't start (Ubuntu 19.04, java 8)

eschercode, 2019-05-03:
## Problem description 
UGS platform crashes a few seconds after startup (before GUI is loaded).

### Steps to Reproduce the Problem
  1. `java -version` output: `java version "1.8.0_211"`
  1. Execute `ugsplatform`
  1. Traceback:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/opt/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Gtk-Message: 13:17:50.456: Failed to load module "canberra-gtk-module"
Inconsistency detected by ld.so: dl-lookup.c: 112: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

```

### Specifications
#### Version
Problem persists on stable build 2.0 linked version, as well as nightly build 2.0 linked version.

#### Operating system
Fresh install of ubuntu 19.04, with java 1.8.211. I've also tried with modern java (java 11), but that has the same problem.

#### Other
I'm finding UGS incredibly sensitive to java dependencies/versions; I can recall only getting a working java config once in the past few years and can't remember what I did. It's disappointing that it's so tightly coupled to whatever netbeans/java trickiness it uses (I don't use java myself since Oracle bought it so I don't know the intricacies). Also, UGS dependancy lists java 8+; perhaps consider removing the "plus" if it doesn't actually work with modern java.

breiler, 2019-05-03:
The error message suggests a missing gtk-module. Try installing it with:
`sudo apt install libcanberra-gtk-module`

https://askubuntu.com/questions/342202/failed-to-load-module-canberra-gtk-module-but-already-installed.

We are waiting on a new release of netbeans that should work with later Java versions. Until then Java 8 is a requirement. I've removed "or later" from the front page in this git repo.

eschercode, 2019-05-03:
Nope, that didn't do it unfortunately:
```WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/opt/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 112: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
```

breiler, 2019-05-03:
Doh!
I know you've checked that you are running Java 8. But could you try running this (It would be interesting of the output if you have Java 11 installed as well):
`sudo update-alternatives --config java`

eschercode, 2019-05-03:
```
There are 2 choices for the alternative java (providing /usr/local/bin/java).

  Selection    Path                                         Priority   Status
------------------------------------------------------------
  0            /usr/lib/jvm/java-11-openjdk-amd64/bin/java   1111      auto mode
* 1            /opt/jre1.8.0_211/bin/java                    0         manual mode
  2            /usr/lib/jvm/java-11-openjdk-amd64/bin/java   1111      manual mode
```
Here's what I get from running `ugsplatform` after changing to Java 11:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/opt/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 112: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
```

PierreAndreNoel, 2019-05-11:
I had the same problem as @eschercode, but I eventually got it to work.

I first ran `sudo apt install openjdk-8-jre` then used `sudo update-alternatives --config java` to select `/usr/lib/jvm/java-8-openjdk-amd64/jre/bin/java`, then ran `sudo apt install libcanberra-gtk-module`, which gave me the same error as @eschercode as of https://github.com/winder/Universal-G-Code-Sender/issues/1228#issuecomment-489071409

I then ran `sudo apt install openjdk-8-jdk` and used `sudo update-alternatives --config javac` to select `/usr/lib/jvm/java-8-openjdk-amd64/bin/javac`, then ran the `ugsplatform/bin/ugsplatform` to get a working UGS.

So in short, you need both JRE 8 and JDK 8 in addition to `libcanberra-gtk-module`.

erichVK5, 2019-05-11:
I am running into similar issues, trying to start after a new maven build of a fresh git clone

for both the classic

[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  0.967 s]
[INFO] ugs-core ........................................... SUCCESS [ 58.457 s]
[INFO] ugs-pendant ........................................ FAILURE [01:02 min]

and the platform versions

[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  1.281 s]
[INFO] ugs-core ........................................... SUCCESS [  8.476 s]
[INFO] ugs-pendant ........................................ FAILURE [ 41.043 s]

I am unfamiliar with maven, being used to coding java in vim and using the CLI for building.

I have done the above recomended stuff, ensuring I am running openjdk-8-jdk, and also have libcanberra-gtk-module installed, and have checked sudo update-alternatives --config java

Regards,

Erich.

breiler, 2019-05-11:
@erichVK5 without the rest of the build log its really hard to help.

erichVK5, 2019-05-11:
It seems the build process fell over before it got to the pendant build part because the GRBL board was not plugged in to the computer. This appears to be mentioned in another "Issue".

After getting over this hurdle, the build process fell over due to a lack of privileges, so I had to run it again as

sudo mvn install 

This seems to have successfully built the project.

neilcsmith-net, 2019-05-28:
Just found this while searching the `dl-lookup` error. Interesting that you also have a NetBeans platform application using JOGL. I'm fairly sure this is a bug being hit during JOGL initialisation - do you do this at startup?  Also interesting to note that I have no problem with OpenJDK 11 from AdoptOpenJDK, but do with the Ubuntu packaged version.  NetBeans 11.0 is out and supports Java 11 well, but may not fix this bug for you.

@PierreAndreNoel is right that the important thing is setting the alternatives for `javac`. The NetBeans launcher looks up `javac` and will use the location of that in preference to the location of `java`.  Setting the `java` alternative or adding `canberra-gtk` won't help (or hinder).  If you see reflection warnings it is never on Java 8.

winder, 2019-05-28:
@neilcsmith-net thanks for stopping by. I've looked at PraxisLIVE in the past for NBP related inspiration (especially when I was trying to bundle the JRE while building the application).

We don't do anything special to initialize JOGL, I believe it happens implicitly when creating a GLJPanel.

neilcsmith-net, 2019-05-28:
@winder but you are creating a GLJPanel as the application starts up?  Interesting you've looked at PraxisLIVE.  That doesn't touch anything JOGL related until you hit the run button - the whole IDE works fine up until doing that, and then immediately has the same error as described here.

It does appear that this is specific to Ubuntu's OpenJDK though. Maybe we need to start bundling JRE's on Linux too?! :grimacing:

winder, 2019-05-28:
Yes, it is created in the context of a default TopComponent, so this crash would happen during startup when JOGL is initialized. Probably if that TopComponent were disabled the behavior would be the same as what you're seeing.

samusamaa, 2019-05-29:
perlun

gamscho, 2019-09-07:
After struggeling for 2 days with the same problem as eschercode I finally found this page on the internet. The classic version 2.0 of ugs was working fine. My Linux platform is Linux Mint Mate v19.2.
The solution of PierreAndreNoel helped me out and all is working ok now.

dzervas, 2019-10-10:
I have the same problem, when I updated from Debian buster to testing...
strace gave some insight, maybe you're able to solve this:

```
stat("/home/dzervas/.ugsplatform/2.0-SNAPSHOT/dev/update/deactivate/deactivate_later.txt", 0x7ffdfdd36da0) = -1 ENOENT (No such file or directory)
stat("/home/dzervas/.ugsplatform/2.0-SNAPSHOT/dev/update/deactivate/to_disable.txt", 0x7ffdfdd36c90) = -1 ENOENT (No such file or directory)
stat("/home/dzervas/.ugsplatform/2.0-SNAPSHOT/dev/update/deactivate/to_uninstall.txt", 0x7ffdfdd36c80) = -1 ENOENT (No such file or directory)
stat("/home/dzervas/.ugsplatform/2.0-SNAPSHOT/dev/var/restart", 0x7ffdfdd37340) = -1 ENOENT (No such file or directory)
```

melvyniandrag, 2019-12-18:
Agree with @PierreAndreNoel . Works after that setup. Surprised that the compiler was needed by this program!
[image]

neilcsmith-net, 2019-12-19:
@melvyniandrag who said it is? The default NB platform launch script will use the location of javac to find where the JRE/JDK is. Nothing more. It makes sense for the NetBeans IDE, but probably be better if platform applications didn't do this.

bianchifan, 2020-02-28:
Same problem here.. Ubuntu 19.10
openjdk-11-jre and openjdk-11-jdk are installed, also libcanberra-gtk-module

On start a couple of warnings are thrown (see 1st post), GCode Sender startup LOGO appears and tells the loaded modules, after a few seconds the main window "Gcode Sender Platform" appears, but grey only inside and crashes after 1 second with the inconsistency failure

breiler, 2020-02-28:
@bianchifan have you tried the version with the bundled java for linux from here?
https://github.com/winder/Universal-G-Code-Sender#downloads

bianchifan, 2020-02-29:
> have you tried the version with the bundled java for linux from here?

No, I ran stable & nightly from the download page.
It seems, I didn't read the README, that version is running fine... thx for the hint!

Nevertheless..the download page tells Java8+ ;)

breiler, 2020-02-29:
Thanks for verifying this! 
We have fallen a bit behind in the documentation... :(

I'm closing this issue as it seems like the bundle is working properly.

closed 2020-02-29 by breiler

bianchifan, 2020-02-29:
[image]

kptjrn, 2020-03-20:
I know this is closed, but just a comment, if someone runs into the same problem I had (Java)... I tried for getting this to run for two days on Ubuntu 18.04 (with my very limit Linux skills)...
The only way I cold get it to work was by completely uninstalling Java with this guide:

- https://askubuntu.com/questions/84483/how-to-completely-uninstall-java

- and then doing exactly what PierreAndreNoel described above, and finally got UGS to run :)

Thank you all!

breiler, 2020-03-20:
@kptjrn if you'd like to run your own version of Java this is probably required as you described.

However, UGS Platform is now packaged  with a bundled Java, you should not need to install it separately anymore.

Just download the linux package from here, unpack and run:
https://github.com/winder/Universal-G-Code-Sender#downloads

tehraio, 2020-07-04:
Just came to say that @PierreAndreNoel method effectively solved the problem for me. I have java11 installed at first, and by installing java8 and selecting it, with the canberra gtk. It now runs fine.

jrrtps, 2020-07-25:
Forget Ubuntu or Linux mint both and others based on ubuntu , I´m using deepin and have no issues maybe due a Debian base...

matthijskooijman, 2020-09-05:
Is there any thought on actually fixing this bug? Just saying "Use this older version of java" is a workaround that helps for now, but is not terribly convenient. Ideally, this would be fixed (in UGS, or java, or wherever the actual cause is) so UGS would just work with any java8+ version...

As for the workaround itself, are there any plans to do a new 2.0 beta/stable release? Or maybe backport the bundled java to the current 2.0 beta/stable release?

breiler, 2020-09-06:
This problem seems to occur with the Java that comes with the linux distribution. If there is something wrong with that Java version there is very little we can do about it in UGS.

The issue is considered closed as UGS now includes Java in the package found here https://github.com/winder/Universal-G-Code-Sender#downloads