winder/Universal-G-Code-Sender #1235: error22 but the G-Code is correct

treki1, 2019-05-11:
My G-Code:
`
(Generated with: DXF2GCODE, Version: Py3.6.7 PyQt5.12.2, Date: $Date: Thu Jan 3 16:29:29 2019 +0100 $)
(Created from file: ....)
(Output format description: G-CODE for LinuxCNC)
(Time: Sat May 11 23:42:18 2019)
G21 (Units in millimeters)
G90 (Absolute programming)
G64 (Default cutting) G17 (XY plane) G40 (Cancel radius comp.) G49 (Cancel length comp.)
G0 Z  20.000 

(*** LAYER: Ebene_1 ***)

(* SHAPE Nr: 2 *)
G0 X 156.984 Y  58.969
M3 S1000
G0 Z   2.000 
F120
G1 Z  -0.100
F1000
G2 X 156.889 Y  59.250 I   0.513 J   0.330
G2 X 156.887 Y  59.300 I   0.624 J   0.050
G2 X 156.887 Y  59.323 I   0.614 J   0.000
.
.
.

`
First, I have to remove the title. Everything in parentheses ()
After starting in UGS 2 the error 22 appears:
"G2X1 56.887Y59.323I0.614J0.000": (error 22) Feed rate has yet been set or is undefined.

In UGS 1.0.9 it works.

Is this a bug in UGS 2+ ?

AbySet, 2019-05-12:
Take a look here: https://github.com/gnea/grbl/issues/255

treki1, 2019-05-12:
Is not possible to add a parameter in the startup from UGS?
The link says, i have to add a line in every gcode.

AbySet, 2019-05-12:
I am just a user like you trying to help people.
The question "Why the error comes only on the 3rd G02 line?" is actualy a good one.
Maybe Winder or breiler can analyse this issue.

AbySet, 2019-05-12:
I just tested you gcode file by removing (comment) G64 line like this:

G21 
G90 
;G64 
G0 Z 20.000

G0 X 156.984 Y 58.969
M3 S1000
G0 Z 2.000
F120
G1 Z -0.100
F1000
G2 X 156.889 Y 59.250 I 0.513 J 0.330
G2 X 156.887 Y 59.300 I 0.624 J 0.050
G2 X 156.887 Y 59.323 I 0.614 J 0.000

And all ran just fine (no errors) on my machine (first ran CHECK mode then run for real).

breiler, 2019-10-27:
Regarding G64, the editor now checks if the command is supported by the hardware and displays a hint. There was an issue with the coordinate parameters containing a space displaying errors which I've fixed.

[image]

I couldn't reproduce the error with the error 22 with missing feed parameter though. I'm closing this, reopen it if the error still exists and provide your current GRBL version.

closed 2019-10-27 by breiler

breiler, 2023-08-18:
@fbonin5491 please open a new issue and attach the sample gcode.