winder/Universal-G-Code-Sender #1237: Stuck on a blank dialog box after launching. Unable to close or continue

moonface1977, 2019-05-19:
## Problem description 
Once launched, it shows the window with the message title "nightly build" but no contents or buttons. I can select the X  to close it but it doesn't close or launchs the UGS. Only way to stop it is from task manager

### Expected Behavior
Launch UGS

### Steps to Reproduce the Problem
1-  open the folder BIN
2- right click and run ugsplatform64

### Specifications
#### Version

AbySet, 2019-05-19:
Please provide us with more informations like your OS, also take a look at UGS files date so we know whitch version of UGS you are using.
Thank you.

JacksCrafts, 2019-05-19:
You might try uninstalling and then reinstalling Java.

moonface1977, 2019-05-19:
Im very sorry. I had a very detailed comment. For some reason it didn't go through.
Looks like it gets stuck on the first warning and I'm unable to do anything.
Here is the info:

Windows 10 Home 64
java Version 8 update 211
UGS nightly April 23, 2019 (also tried other versions)

Steps I took to fix:
*Unistalled UGS completely, deleted the files under Appdata (including roaming). Restarted PC
*Installed UGS again. Restarted PC
*Uninstalled Java completely and installed again. Restarted PC
*run UGS from command line, same results. Please see attached as it looks like stops after the second 
  message and stays there. (I've see other screenshots and it doesn't get that far)
*Run UGS in compatibility mode
*Updated Video drivers.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/3195707/messages.log)

moonface1977, 2019-05-19:
Here are the screenshots.

[image]
[image]

AbySet, 2019-05-19:
Did you deleted a folder called ".ugs" under \users\username\ ?
complete path: c:\users\username\.ugs

Also you may try a classic version just to see if it's the interface or something else causing the issue and can help to go to the source of the problem.
I do use win10 x64 and same version of ugs but everyting is working fine to me.

Some antivirus can hold the execution of programs when they are newly installed for about 15 seconds before they let them run (avast do that).

moonface1977, 2019-05-20:
It was the antivirus (Comodo). Even with all the features disabled it was creating an issue. I attempted to create an exception with no results. Completely uninstalled it and Voila!
THANK YOU SO MUCH FOR YOUR PROMPT HELP!
Once i figure how to create the proper exception I will post it here for future reference

breiler, 2019-05-20:
If you got the time we'd appreciate if you could write a couple of lines about your findings here:
https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#antivirus-software

Thanks!

closed 2019-06-03 by breiler