winder/Universal-G-Code-Sender #1238: new jog controller wont accept X axis shortcuts

jahnj0584, 2019-05-20:
alt up/down ant alt pg up/down work for the Y and Z axis.

alt left/right do not effect the X axis.

yes the keybinds are set to what i want it to do. worked in the april version. now trying the 5/19 platform with the diagnal jog options

breiler, 2019-05-22:
I don't have a Windows machine to try this out. But I experience similar problems on Mac OSX where the operating system already have stuff mapped or won't let me assign certain key combinations.

I've just tried different combinations and finally found combinations working...

It's weird though that shortcuts stopped working, did you perhaps upgrade Windows or installed other software that may interrupt certain key combinations?

jahnj0584, 2019-05-22:
Nothing changed. If I use the old version it still works.

breiler, 2019-05-22:
Thanks, I'll try to look into it!

breiler, 2019-08-12:
I can't reproduce this, but I found an issue where some input boxes (still) steal focus making the shortcuts inactive... This could be what you are experiencing.

I'm working on a potential fix/workaround for this

breiler, 2019-08-14:
Once this has been built you can try the new button and see if it helps, otherwise we'll open this again.

closed 2019-08-14 by breiler