winder/Universal-G-Code-Sender #1239: Probe module download??

tkms002, 2019-05-27:
I have UGS 2.0 installed.  In the plugins list I have PROBE HELPER but I cannot find the PROBE MODULE that so many vids reference.  It seems a lot more friendly than the PROBE HELPER that I have.
Probe helper seems to work but it probes Z very slow, stops when it touches and does no retract.
How do I get the PROBE MODULE?
Thanks

JacksCrafts, 2019-05-28:
If you go to help and click on About you can copy the version you have.

Product Version: Universal Gcode Platform **20190423**
Java: 1.8.0_211; Java HotSpot(TM) 64-Bit Server VM 25.211-b12
Runtime: Java(TM) SE Runtime Environment 1.8.0_211-b12
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)

2019 April 23

[image]

tkms002, 2019-05-28:
So I can get the PROBE MODULE downloaded from the help?

JacksCrafts, 2019-05-28:
No. I was just trying to see what version of UGS you have. By going to Help and then clicking on About, a window will pop up showing what version of UGS you are using. Just saying UGS 2.0 is not enough info.
Are you using the latest version. 04-23-2019?

The probe module:
I did not have to download anything extra. It came pre-installed.

tkms002, 2019-05-28:
this is what I get when I hit help/about
[image]

winder, 2019-05-28:
Looks like you're running a really old version, try upgrading to the latest build.

tkms002, 2019-05-28:
is there a way to upgrade without loosing my settings?

tkms002, 2019-05-28:
also  how do you determine if you should run ugsplatform or ugsplatform64?

tkms002, 2019-05-28:
I got it working!!
Thanks for your help.

closed 2019-07-02 by breiler