winder/Universal-G-Code-Sender #1241: UGS Platform fails to open after Java license update (Apr 16, 2019)

landbrad, 2019-06-02:
I've been using UGS Platform2.0 nightly build for over a year with java 8 on Win 7 / 64bit.  I allowed Java to update to build 1.8.0_211-b12 and now UGS will not open. Opening dialog box advances to the point where it reads "Opening Modules" then disappears.  I've tried deleting the UGS log files and DLing new UGS zip files to no avail.....
Any suggestions? Anyone else having a similar issue?

  What hardware are you using. Example:
  GRBL 1.1f
  
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/3244458/messages.log)

AbySet, 2019-06-02:
I had this issue a year ago, try these two steps:
- Update UGS Platform as well.
- Delete this folder to have a clean installation (Your UGS settings will be deleted):
   c:\users\<username>\ .ugs       
   c:\users\<username>\AppData\Roaming\ .ugsplatform
   - _(Without spaces after \ in both lines, i had to add them)._

You can also check here for Troubleshooting overview:
https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#antivirus-software

landbrad, 2019-06-02:
Thanks AbySet but I tried that.
I'm almost certain this is related to the recent Java update on April 16, 2019...

AbySet, 2019-06-02:
I am using it and i have no problems at all.
Maybe @winder or @breiler can take a look and help you, they have much more knowlege than me, just be patient, i'm sure they will take a look at your post.
Wish you good luck.

breiler, 2019-06-03:
Sorry, I don't have a Windows machine to try reproducing the problem on. 

But from the log messages. it really looks like the settings file is corrupt. Would you mind attaching your settings file located in c:\users\username\ .ugs?

landbrad, 2019-06-03:
> Sorry, I don't have a Windows machine to try reproducing the problem on.
> 
> But from the log messages. it really looks like the settings file is corrupt. Would you mind attaching your settings file located in c:\users\username\ .ugs?

The .json file ?

landbrad, 2019-06-03:
Hmmmm
The only files in the .ugs folder are .json files which aren't supported here for uploading.

JacksCrafts, 2019-06-03:
You might try uninstalling java and then go download it again.

landbrad, 2019-06-06:
Reinstalled Windows and started from scratch. Back in business now...

closed 2019-06-12 by breiler