winder/Universal-G-Code-Sender #1242: UGS not connecting with new computer to exisiting working xcarve which works with other computer.

PatrickGazarkiewicz, 2019-06-05:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

PatrickGazarkiewicz, 2019-06-05:
I am having an issue connecting UGS on my newer Dell Windows 10 laptop.  I have updated/loaded and reloaded Java 8.  I open the nightly platform UGS .  Unzip no problem click on bin and create shortcut.
I see the version is new.  When I connect to the Xcarve control box I click connect and cannot get it to connect.  I switch computers click on UGS and connect.  All goes green and I have full control.
I have uploaded a gcode project I know works and I get a grbl has not completed or is still loading?
Ive tried loading older platforms.  No problem loading but still no connection when the connect button is clicked on.
Any help would be great.

Many thanks

Patrick

PatrickGazarkiewicz, 2019-06-05:
Have unloaded and reloaded new JAVA.  Doesn't seem to help.

winder, 2019-06-05:
Make sure you have any required serial drivers installed on your new computer. If this page is correct for your machine, it looks like you just need the standard arduino drivers: http://x-carve-instructions.inventables.com/xcarve2015/step16/

AbySet, 2019-06-05:
Those are two different issues:
- (I am having an issue connecting UGS on my newer Dell Windows 10 laptop)
- (I have uploaded a gcode project I know works and I get a grbl has not completed or is still loading?)
The first one is connection issue, the second one is a file loading issue.
Could you be more specific on whitch one is the problem ?

I hope i am not missunderstanding your post.

PatrickGazarkiewicz, 2019-06-06:
Sure:
I’m sorry for bluring the question.
I just can’t get UGS to connect to my
Xcarve CNC with my new computer.  When I click the connect Icon nothing happens.
My other laptop is only a year old so they are both relatively new.  I also run laser GRBL
And both work fine on the older windows 10 /64 laptop.
  I have downloaded ardino GRBL, Java 8 and UGS and all go well but the new computer just won’t connect.
I did try again tonight to remove and reload the GRBL but no go.  
Many thanks
P

Patrick Gazarkiewicz, M.A.,
 F.F./Advanced E.M.T

AbySet, 2019-06-06:
Try to get to [Tools] -> [Options] -> [UGS] Tab -> [Sender options] Tab and play arround "connection driver" by changing it to JSerialCom or JSCC.
Also did you installed the controler serial drivers as @Winder has recommanded ?

On the top window of UGS under the menu bar:
- Is Firmware set to GRBL or something according to your machine ?
- Did you refresh the Port ? Click on the double arrow button.
- Is the baudrate set to 115200 ?

Other thinks to check:
- USB cable still good.
- USB port is enabled in the bios of your new laptop.
- On windows, check the device manager for any usb device where its properties can show power managment option and disable "Allow PC to power off this device for energy saving".
- Check the serial ports parameters in device manager if they are set to:
   Parity: none
   Stop bit: 1
   Flow control : none _(this may be different word, my pc is frenchy)_

PatrickGazarkiewicz, 2019-06-07:
Thank you so much for all of the great suggestions.  I did check all of my usb
Ports and took them off of sleep option.
I did have the Java, GRBL and UGS configured properly.  I had a bit more time and went back to the inventables xcarve site
And tried using their easel program.
I found another help section I missed and it recommended for the Xcarve GRBL that I needed to load a FTDL file??
I did and now have a great connection!!
I noticed my stepper motors are moving but very slowly I have them entered to move an inch in  X Y Z and they take several minutes to move that far??

I guess I will tinker with the newer UGS screen and see what’s slowing things down.
Thank you to everyone who helped me solve
My problem.  I teach HS AP Chemistry and
Know how important  and wonderful it is to have people not giving up on a problem I can’t say thank you
Enough.
Patrick 

Patrick Gazarkiewicz, M.A.,
 F.F./Advanced E.M.T

AbySet, 2019-06-07:
Thank you @PatrickGazarkiewicz for replying how you did fix your issue, i am just a user like you and i believe that more i help more i learn, i am one of those who are in love with cnc machines.

So far so good i will keep in mind that FTDI driver has resolved XCarve connetion issue.

BTW dont forget to reactivate "Allow PC to power off this device for energy saving" for USB devices since it was not the one causing the issue and your PC is a laptop and may need it more than a desktop PC.

Enjoy your carving.

closed 2019-06-07 by winder

PatrickGazarkiewicz, 2019-06-07:
Many thanks :)

Patrick Gazarkiewicz, M.A.,
 F.F./Advanced E.M.T