winder/Universal-G-Code-Sender #1245: Install on Raspberry Pi

cvboom, 2019-06-13:
## Problem description 
Getting error on install of UGS on Raspberry Pi.

### Expected Behavior
Normal install

### Actual Behavior
All seems fine up to:

[INFO] ------------------------------------------------------------------------
[INFO] Building ugs-pendant 2.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) @ ugs-pendant ---
[INFO] Installing node version v10.15.0
[INFO] Downloading https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz to /home/pi/.m2/repository/com/github/eirslett/node/10.15.0/node-10.15.0-linux-x86.tar.gz
[INFO] No proxies configured
[INFO] No proxy was configured, downloading directly
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  2.276 s]
[INFO] ugs-core ........................................... SUCCESS [ 20.530 s]
[INFO] ugs-pendant ........................................ FAILURE [  5.831 s]
[INFO] ugs-cli ............................................ SKIPPED
[INFO] ugs-classic ........................................ SKIPPED
[INFO] ugs-platform-parent ................................ SKIPPED
[INFO] ugs-platform-branding .............................. SKIPPED
[INFO] ugs-platform-ugslib ................................ SKIPPED
[INFO] ugs-platform-ugscore ............................... SKIPPED
[INFO] ugs-platform-visualizer ............................ SKIPPED
[INFO] ugs-platform-plugin-workflow ....................... SKIPPED
[INFO] ugs-platform-gcode-editor .......................... SKIPPED
[INFO] ugs-platform-surfacescanner ........................ SKIPPED
[INFO] ProbeModule ........................................ SKIPPED
[INFO] DowelModule ........................................ SKIPPED
[INFO] GcodeTools ......................................... SKIPPED
[INFO] ugs-platform-plugin-jog ............................ SKIPPED
[INFO] ugs-platform-plugin-setup-wizard ................... SKIPPED
[INFO] ugs-platform-welcome-page .......................... SKIPPED
[INFO] ugs-platform-app ................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 36.021 s
[INFO] Finished at: 2019-06-13T12:43:58+02:00
[INFO] Final Memory: 22M/53M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal com.github.eirslett:frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) on project ugs-pendant: Could not download Node.js: Got error code 404 from the server. -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :ugs-pendant

### Steps to Reproduce the Problem
git clone https://github.com/winder/Universal-G-Code-Sender
cd Universal-G-Code-Sender/ 
./run_platform.sh

### Specifications
Going to: https://nodejs.org/dist/v10.15.0

https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz - Not found!

#### Version
Current version on the web.

#### Operating system
 Linux on RaspberryPI
Debian version 9.9

Raspbian GNU/Linux 9 (stretch)

#### Platform
Raspberry Pi 3 Model B +

#### Other

breiler, 2019-06-13:
I think that this might be the issue: https://github.com/eirslett/frontend-maven-plugin/issues/704

I should probably add an option to disable the pendant ui build... That way you could at least build it.

cvboom, 2019-06-15:
Any idea on when you would be able to add this option? 

Kind regards, Chris

scottbarry2001, 2019-07-26:
Hello,
Do you know if this was fixed or if the option was added?  I am running into the exact same issue.

piedes, 2019-07-27:
Hi,
Have the same issue. Any news?
It is possible to disable ugs-pendant manually?

Regards

breiler, 2019-07-29:
I've added a profile in the build script for the pendant UI which is enabled by default. 

To disable the   module you can run the following:
`mvn install -P '!create-pendant-web-ui'`

Sorry for the  inconvenience 😞

closed 2019-07-29 by breiler

wooddragon48, 2019-08-13:
I'm having the same issue and the !create-pendant-web-ui profile does not work either.
I tried with the current files in the repository, and the build fails on the core.  
I see that breiler closed this issue, but I would like to know if there is a fix for this issue?

cvboom, 2019-08-21:
I used the mvn install -P '!create-pendant-web-ui' but still getting the same error. I verified that profile is used. 

So, not solved :-(

[INFO] Installing node version v10.15.0
[INFO] Downloading https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz to /home/pi/.m2/repository/com/github/eirslett/node/10.15.0/node-10.15.0-linux-x86.tar.gz
[INFO] No proxies configured
[INFO] No proxy was configured, downloading directly
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  1.233 s]
[INFO] ugs-core ........................................... SUCCESS [01:08 min]
[INFO] ugs-pendant ........................................ FAILURE [  3.137 s]
[INFO] ugs-cli ............................................ SKIPPED
........etc

Regards,
Chris

breiler, 2019-08-21:
That's strange, and you are certain that you've got the latest code?

What maven version are you running?
`mvn -version`

cvboom, 2019-08-21:
Output from mvn -version : 

Apache Maven 3.3.9
Maven home: /usr/share/maven
Java version: 1.8.0_65, vendor: Oracle Corporation
Java home: /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre
Default locale: en_GB, platform encoding: UTF-8
OS name: "linux", version: "4.19.42-v7+", arch: "arm", family: "unix"

CMaat, 2020-11-16:
almost at the end of the installation I have the following problem
(RPB 4, 1 GB RAM)

[INFO] ugs-platform-ugslib ................................ SUCCESS [ 25.128 s]
[INFO] ugs-platform-ugscore ............................... SUCCESS [ 15.104 s]
[INFO] ugs-platform-visualizer ............................ SUCCESS [ 12.366 s]
[INFO] ugs-platform-plugin-workflow ....................... SUCCESS [  5.724 s]
[INFO] ugs-platform-gcode-editor .......................... SUCCESS [ 14.512 s]
[INFO] ugs-platform-surfacescanner ........................ SUCCESS [  5.583 s]
[INFO] ProbeModule ........................................ SUCCESS [  9.788 s]
[INFO] DowelModule ........................................ SUCCESS [ 31.259 s]
[INFO] GcodeTools ......................................... SUCCESS [  4.918 s]
[INFO] ugs-platform-plugin-jog ............................ SUCCESS [ 24.789 s]
[INFO] ugs-platform-plugin-setup-wizard ................... SUCCESS [  8.790 s]
[INFO] ugs-platform-welcome-page .......................... SUCCESS [  5.935 s]
[INFO] ugs-platform-plugin-joystick ....................... SUCCESS [ 32.079 s]
[INFO] ugs-platform-plugin-cloud-storage .................. SUCCESS [ 31.576 s]
[INFO] ugs-platform-plugin-toolbox ........................ SUCCESS [  7.385 s]
[INFO] ugs-platform-app ................................... FAILURE [ 24.220 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  06:11 min
[INFO] Finished at: 2020-11-16T18:28:18+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.netbeans.utilities:nbm-maven-plugin:4.4:standalone-zip (default-standalone-zip) on project ugs-platform-app: : MojoExecutionException: Problem creating zip: Execution exception: Java heap space -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :ugs-platform-app

how can i fix this?

Regards,
Maat

breiler, 2020-11-16:
@CMaat is there a reason you are compiling it on the raspberry instead of downloading the prebuilt binary:
https://github.com/winder/Universal-G-Code-Sender#downloads

You could try setting the environment variable before building: `export MAVEN_OPTS=-Xmx1024m`