winder/Universal-G-Code-Sender #1246: Z probe error

khturnings, 2019-06-13:
I get this error when trying to run a Z probe.

I'm running on
Product Version: Universal Gcode Platform 20190423
Java: 1.8.0_152; Java HotSpot(TM) 64-Bit Server VM 25.152-b16
Runtime: Java(TM) SE Runtime Environment 1.8.0_152-b16
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)

### Expected Behavior
I think this is supposed to touch the plate, and then pull up a safe distance plus set Z to zero allowing for the thickness to the plate.

### Actual Behavior
Z axis drops down and touches the touch plate then it produces the error shown in the picture

### Steps to Reproduce the Problem
1.  Move the Z axis within 10mm of touch plate
2. Click in 'Initialize' 

#### Platform

[image]

AbySet, 2019-06-13:
Make sure $6 is set to zero.
$6 = 0

JacksCrafts, 2019-06-14:
Move the Z axis within 10mm of touch plate

Try to move the Z within 5 MM  of the touch plate to make sure you are not going over the limit you have set.

khturnings, 2019-06-14:
It turns out I was using an older version of UGS from September 2018. I upgraded to the new version April 2019 and run probe again and all worked fine i.e. no errors. 

It touched, retracted and the slowly re-touched. however Z was set to the distance it retracted to. I then needed to change the Z value to Z - retract?
Is that right?

closed 2019-06-25 by khturnings