winder/Universal-G-Code-Sender #1247: raspberry pi visualization while using VNC

khturnings, 2019-06-25:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

khturnings, 2019-07-17:
Anyone have a solution for this, is it an issue with the new Raspbian Buster or another issue?
thanks

winder, 2019-07-17:
@khturnings you could give this a shot: https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#activate-opengl-driver

khturnings, 2019-07-17:
@winder I used these instructions on a clean buster build - I'm using a headless setup via VNC.
To use VNC it requires the /opt/vc and I guess without removing this directory the opengl driver does not get fully enabled.
Not sure what the solution would be?

winder, 2019-07-17:
Ah, sorry about not reading your entire bug report. This is a long time issue and I assumed this was the version that we've seen on the issue board many times before.

I don't know a lot about this, but it looks like you need to install videocore6. After you reboot, does it re-create the /opt/vc directory?

I did find an interesting comment while running a quick keyword search on this issue... [on this page](https://github.com/raspberrypi/documentation/blob/master/glossary/README.md):
```
On the Pi 4, a newer and faster version called VideoCore6 (VC6) is used.
```

Seems like the Pi 4 should work out of the box!

khturnings, 2019-07-17:
@winder Interesting - The /opt/vc directory is not re-created when rebooted. 
Shame I am using a Pi 3 - someone else suggested reverting back to a Raspbian Stretch and then following your instructions.
I will give it a try and update this thread.

BusbyGT, 2019-08-16:
I came across this issue and tested this on a Raspberry Pi 4 running Raspbian buster.

1) Running headless and renaming the vc folder caused the VNC issue described here
2) Visualizer on platform 4/23/19 doesn't work
3) Changing GL driver in raspi-config doesn't fix the Visualizer (i'm using the new v3d by default anyways). vc4 driver is still being used as well. vc6 is the chip itself from what I've researched.
4) Java 1.8 and OpenJDK v11 both had same results
5) Verified other GL programs work

I might be able to try it on a display when my new case arrives.

khturnings, 2019-08-16:
Hi All
A friend of mine has a pi4 and I got him to test UGS and following the instructions on wiki he too confirmed that the visualizer did not work.

BUT all is not lost, he came up with a partial solution for running UGS as in headless mode.

1. Follow the Wiki but do not rename the /opt/vc directory to /opt/vc.old
2. Open a VNC session - this will now work ok.
3. Now rename /opt/vc to /opt/vc.old
4. Start UGS by running ugsplatform in normal way.

Now UGS works as does the Visualizer.
If running with a screen this is not required I believe.

If I get time I might write a small script to automate the process above and then rename /opt/vc.old back to /opt/vc when UGS closes. If you don't then you will not be able to VNC into the machine a second time and will need to ssh into machine to rename the directory.

Hope this helps

khturnings, 2019-08-16:
Upon unloading UGS I saw a number of these messages - i'm not sure if these will be an issue. I have not run it while linked up to my CNC.

MESA-LOADER: failed to retrieve device information
libGL error: MESA-LOADER: failed to retrieve device information

khturnings, 2019-09-08:
Above error is still not tested but I did write this simple script to automate 

> Hi All
> A friend of mine has a pi4 and I got him to test UGS and following the instructions on wiki he too confirmed that the visualizer did not work.
>
> BUT all is not lost, he came up with a partial [...]

Here is the script I wrote to automate this - 

#!/bin/sh
sudo mv /opt/vc /opt/vc.old
sudo /home/pi/ugsplatform/bin/ugsplatform
sudo mv /opt/vc.old /opt/vc
echo "Finished"

TheNetStriker, 2020-03-12:
@khturnings  I found out that the vc file only has to be renamed as long as the application is starting up. After that it can be renamed back and everything seems to work. I've added some lines to you script to automate this:

#!/bin/sh
sudo mv /opt/vc /opt/vc.old
sudo /home/pi/ugsplatform/bin/ugsplatform &
sleep 30
sudo mv /opt/vc.old /opt/vc

If the application needs more than 30 seconds to start up just increase the sleep timeout.

sepiroth887, 2022-11-15:
pretty old thread, but i stumbled here after trying without luck to get the visualizer working on pi 4 bullseye amd64 install with 2.0.12. 

Turns out the switch to new openGL handler since this commit [d432d5e](https://github.com/winder/Universal-G-Code-Sender/commit/d432d5ec3891f8347a457e1ce13c0b4a244a9c37) made everything work :)