winder/Universal-G-Code-Sender #1248: UGCS GUI Not Opening

NicholasTomas, 2019-06-27:
Hey guys so I have been trying to download UGCS to run with my DUET3D and CNC machine. I've downloaded the new version of Java as well as the correct drivers on RepRap for the Duet controller. For some reason every time I try to open up the platform on my Windows 10 computer, this happens: 

[image]

It never leaves this point and nothing happens afterwards. I have used multiple computers and trying it now with a MAC. I am not sure why it would be doing this. I've followed all the steps and have tried restarting the computer a bunch of times. Please help!!

Thanks,

Nick

AbySet, 2019-06-28:
Where did you put "ugsplatform" folder ? 
How are you running the program ? (i'm asking this because i don't see cmd window when i run mine).

I suggest to put the "ugsplatform" folder somewhere in user files location like desktop and open "ugsplatform\bin\" folder and create a shortcut from one of those two exe files depending on your system.
(i'm using windows 10 also and all is good for me).

_(I'm just a user like you, a solution feedback is always welcome to me and can help others)_

NicholasTomas, 2019-07-01:
Is it supposed to work with windows 7? We have a bunch of different computers. I meant to say I've tried Windows 7 and earlier but not 10. I will try your recommendation now and let you know how it goes!

Thanks.

breiler, 2019-08-06:
@winder I think we need a new build of the classic edition. I get the same error when trying to open the classic edition. =(

winder, 2019-08-20:
@NicholasTomas please give the new build a try when you have a chance: http://bit.ly/2HhJIir

AbySet, 2019-08-22:
Just for info: I downloaded from provided link, ran it and it worked on Windows 10 x64 May2019 Version.

(Extracted folder stands on desktop).

bobdewar, 2019-09-25:
After reloading Windows 10 on my laptop I can no longer open UGS Platform.  I can start UGS Classic but the window closes when I try and start the Visualizer.   I uninstalled and reinstalled Java 8 and there is not change.  I can run UGS Classic and Platform on my desktop PC with no problem.

I copied the text below from the log file.  My apologies if this is not the correct process to request support. 

# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x697592de, pid=1000, tid=0x0000085c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_221-b11) (build 1.8.0_221-b11)
# Java VM: Java HotSpot(TM) Client VM (25.221-b11 mixed mode windows-x86 )
# Problematic frame:
# C  [ig9icd32.dll+0x392de]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

breiler, 2020-04-10:
Use the builds from https://github.com/winder/Universal-G-Code-Sender#downloads
These now comes with a bundled Java and the problem should no longer exist.

closed 2020-04-10 by breiler