winder/Universal-G-Code-Sender #1249: Confirmation of "Reset Zero"

AngeloQABR, 2019-07-04:
Hi!
"Reset Zero" button is just near "Return Zero" button... I know, is my problem but I clicked "Reset Zero" button for two times when I wanted return to zero.
Is it possible create a message for confirmation in "Reset Zero" button? It would have saved me on the two jobs I lost.
Thanks!

AbySet, 2019-07-06:
This happend to me too, but since it is not a bug i did not post it.
I should say i totaly agree with you @AngeloQABR , any improuvement will be welcome or at least colored text in red or orange but confirmation is better.
Thanks to you and to everyone working on this project.

jahnj0584, 2020-02-02:
Quickest solution is to make these commands as a seperate macro, which you can easily re-order and seperate from each other

TekMason, 2020-02-02:
+1 for this feature built-in.

breiler, 2020-10-19:
It is now possible to reorder the buttons in the toolbox window:
https://github.com/winder/Universal-G-Code-Sender/wiki/Usage#toolbox

closed 2020-10-19 by breiler