winder/Universal-G-Code-Sender #1251: UGS classic failure to open on linux mint 19.1

loonster, 2019-07-22:
## Problem description 
UGS classic has been working on Linux Ubuntu 16 for two years.  Recently installed Linux Mint 19.1 and then installed Oracle's JRE version 1.8.0_221.  UGS fails to open, giving an error message, seen below.  This occurs with the Dec. 2018 version of UGS and also with the July 22 release of version 2.

tim@tim-Latitude-E6440:~$ java -version
java version "1.8.0_221"
Java(TM) SE Runtime Environment (build 1.8.0_221-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.221-b11, mixed mode)
tim@tim-Latitude-E6440:~$ cd Documents
tim@tim-Latitude-E6440:~/Documents$ java -jar -Xmx256m UniversalGcodeSender.jar
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: JogPanel
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: JogPanel
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MacroPanel
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: CommandTextArea
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: ConfigureFirmwareAction$$Lambda$46/710239027
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: Initializing firmware... ...
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/
Jul 22, 2019 12:37:10 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/test_2.json
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/test_1.json
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/grbl.json
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/smoothieboard.json
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.utils.FirmwareUtils initialize
INFO: /resources/firmware_config/tinyg.json
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: MainWindow
Jul 22, 2019 12:37:11 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MainWindow
Exception in thread "main" java.lang.NoClassDefFoundError: com/willwinder/universalgcodesender/utils/KeepAwakeUtils
	at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:140)
	at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:223)
Caused by: java.lang.ClassNotFoundException: com.willwinder.universalgcodesender.utils.KeepAwakeUtils
	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)

loonster, 2019-07-22:
This problem is not specific to linux mint 19.1, as I tried opening UGSclassic latest version on a macintosh running OS X 10.11.6 with oracle java 1.8.0_221.  Same result:  

Exception in thread "main" java.lang.NoClassDefFoundError: com/willwinder/universalgcodesender/utils/KeepAwakeUtils

loonster, 2019-07-22:
UGSclassic 12/18/2018 version with oracle java 1.8.0_31 works perfectly.

winder, 2019-07-23:
Did you upgrade from an older version of UGS? This is related to some
refactoring that was done recently, you might need to clear out your var
directory:
https://winder.github.io/ugs_website/guide/troubleshooting/#property-files

loonster, 2019-07-23:
I’ll check there, but meanwhile FWIW here is a chart I made of the problems encountered yesterday:

It sure would be nice if all these java incompatibilities could be resolved, and in particular if UGS would work with open source java

loonster, 2019-07-23:
Testing with two versions of UGS classic today; preferences files attached:

breiler, 2019-08-09:
Closing as its a duplicate of #1248

closed 2019-08-09 by breiler