winder/Universal-G-Code-Sender #1254: Can't launch UGS Platform to launch using OpenJDK12.0.2 on Windows

marciot, 2019-08-11:
I am trying to run UGS using the [GPL version of OpenJDK](https://jdk.java.net/12/) on Windows. I installed the OpenJDK files to "C:\Program Files\OpenJDK\jdk-12.0.2" and then added "C:\Program Files\OpenJDK\jdk-12.0.2\bin" to the system PATH and created a JAVA_HOME environment variable with the value "C:\Program Files\OpenJDK\jdk-12.0.2"

From the command prompt, I can run "java" and I get:

```
C:\Users\test8>java --version
openjdk 12.0.2 2019-07-16
OpenJDK Runtime Environment (build 12.0.2+10)
OpenJDK 64-Bit Server VM (build 12.0.2+10, mixed mode, sharing)

```

However, when I double-click on "ugsplatform64", I get "Cannot find Java 1.7 or higher".

Am I doing something wrong? Does UGS Platform require the non-GPL Java install?

winder, 2019-08-11:
If you run UGS from the command line does it work? Perhaps you need to restart after installing OpenJDK?

If none of that works you can configure the path to the java runtime directly by following directions linked to in #885

marciot, 2019-08-11:
@winder: Thank you for the quick reply! Restarting made no difference. Running from the command line gives me the same error about Java not being found, with the following in the console:

```
C:\Users\test8\Downloads\ugsplatform-2.0.0-stable\ugsplatform\bin>

The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
```

I then modified `ugsplatform.conf` with the following line:

```
# default location of JDK/JRE, can be overridden by using --jdkhome switch 
#jdkhome="C:\\Program Files\\OpenJDK\\jdk-12.0.2"
```

This changed the behavior somewhat. instead of the error about not being able to find Java, I instead get a `Netbeans Platform 8.1` splash screen for a moment, but then nothing else happens. The following is printed in the console:

```
C:\Users\test8\Downloads\ugsplatform-2.0.0-stable\ugsplatform\bin>

The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/C:/Users/test8/Downloads/ugsplatform-2.0.0-stable/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

So, no dice with OpenJDK so far.

breiler, 2019-08-12:
@winder I think we still need to merge #1243 to have it working on Java 9 and later.

marciot, 2019-08-17:
@winder, @breiler: I noticed that #1243 had been merged, so I tried the nightly build today. While it did not work with an unmodified "ugsplatform.com," it did work once I modified the "jdkhome" variable:

```
# default location of JDK/JRE, can be overridden by using --jdkhome switch 
jdkhome="c:\\Users\\USERNAME\\Downloads\\openjdk-12.0.2_windows-x64_bin\\jdk-12.0.2"
```

The procedure for setting this path was not obvious until I asked my question here. Perhaps it makes sense to add a note to the downloads page, for people who want to use OpenJDK?

breiler, 2019-08-17:
The nightly builds are actually not up and running yet so I think that you're still running an older version? And I'm surprised that you got it working with a later JDK. 

As soon as @winder get the nightly builds up and running (or makes a new build), no modifications or extra parameter tinkering should be necessary.

marciot, 2019-08-17:
That’s peculiar. I did notice the zip file I got when I clicked nightly was labeled “stable” but it was bigger than the zip file I get when I clicked the link to the stable release. So I don’t know what version I am running, but the one linked from the nightly link seems to work *shrug*

marciot, 2019-08-17:
@breiler: It says “version 2.0 [nightly] / Apr 23, 2019” on the titlebar of the main window.

winder, 2019-08-20:
@marciot please give the new build a try when you have a chance: http://bit.ly/2L2v9k6

marciot, 2019-08-21:
> @marciot please give the new build a try when you have a chance: http://bit.ly/2L2v9k6

@winder: With that version, it still says I don't have the correct Java, but once I change the jdkhome then it starts up. So it behaves exactly like the other nightly release I had tried (but this one says Aug 20, 2019 in the title bar)

breiler, 2019-11-14:
It is a bit tricky to get it running on Windows, you need to change the jdkhome configuration as you mentioned. I'll see if we can create a windows bundle that includes the JDK.

breiler, 2019-11-14:
Still need some testing, but there is a build available at: https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT.win.zip

breiler, 2019-11-16:
I've tested this on Windows 10. I'd be interested if anyone has the time to try it out.

Download, unzip and run the exe in the bin-directory. There should be no need for installing Java or changing configuration files.

MikeMitterer, 2020-01-10:
I had the same problem with openJDK 13.0.1 on Windows 10

But thanks to @marciot I found etc/ugsplatform.conf and modified 
`jdkhome` accordingly - now it works.

breiler, 2020-01-12:
Closing as there is now a window download with a bundled Java.

closed 2020-01-12 by breiler

rmfolsom, 2020-05-31:
I see that this is closed, but seems to still be a problem, at least for me.  I am running Windows 10.  Installed Java.  Both have automatic updates so are current.  Downloaded the latest nightly build (ugs-platform-app-2.0-SNAPSHOT.zip) and unzipped to C:\Program Files\ugsplatform.  Go to \bin and double click ugsplatform.exe which works.  ugsplatform64.exe errors with a message: Cannot find Java 1.8 or higher.  Wondering if I should even care about running the 32-bit vs. the 64-bit version.  If there is some script or other command line that is required, might be nice to provide guidance where it can be found easily.

breiler, 2020-05-31:
@rmfolsom download the Windows-version from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

It comes with a bundled Java, so there's no need to install it separately.

rmfolsom, 2020-05-31:
Joacim,

It works 😀  Thank you for the very quick response.

Cheers,
Randy