winder/Universal-G-Code-Sender #1257: UGS Crashing daily

mjsfabshack, 2019-08-12:
## Feature request
 The classic version worked great until the Sunday.

## Problem description 
 The download deletes itself everyday and have to reinstall everyday.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.Click to open program and its no longer there.
  1.
  1.

### Specifications
#### Version

#### Operating system
Windows 10

#### Platform
GRBL
[image] This version worked fine just had an issue with bigger files not loading up. I use this to run a 4x4 CNC Plasma Table

#### Other

breiler, 2019-08-13:
Which version och the UGS Classic are you using?

How did you install the software (step by step)?

How are you starting the software, through the .bat-script or by double-clicking the jar-file?
(http://winder.github.io/ugs_website/guide/classic/#how-to-run)

What do you mean by getting deleted, is the jar-file removed?

The version for UGS Platform that you made a screen shot of is old. Try upgrading to the latest nightly build.

mjsfabshack, 2019-08-13:
I downloaded the latest stable version that I could. i got it to work yesterday its 2.0 that New Gen platform. I'm at work but I can check this all this evening to be sure about it. 

I clicked download off GitHub
Extracted files like the instructions told me to do
then clicked the jar file

I downloaded this a few times before it would actually run trying different versions yes i deleted everything before a reinstall. 

When i would close out the program when I was done. If I went back in to reopen it all the files would be gone even doing a search across my computer nothing.

The one in the screen shot was Classic Platform nightly build.

I need to try this version that is actually working right now to see if it will open a big file. I had a sign I was doing at 8 inches and then going to make one at 10 inches. it would never load up in the classic platform shown in the screen shot above.

BallscrewBob, 2019-08-31:
Large files cause all sorts of seemingly random issues.
From exceptionally slow load times to failing to end when finished. These can be a few minutes or even hours.
As for files going missing I have had that here too when looking for the log files from UGS.

Stepping back versions seems to be the only method I have found so far to reduce load times for a file and even then I can repeatedly get the failing to finish.
This despite changing the buffer to a larger size as was recommended.

So far it appears as it did once before to be a limit to the buffers and being over run in some form of memory leak.

Bob.

breiler, 2020-12-08:
The memory issue have been solved. Please upgrade to the latest stable version to see if this fixes it. Otherwise feel free to reopen the issue.

closed 2020-12-08 by breiler