winder/Universal-G-Code-Sender #1262: Improve support for TinyG controllers

breiler, 2019-08-22:
## Feature request
I've just got a TinyG board and hooked it up to UGS nightly. There where some basic stuff that's not working at the moment, like the initialization of the controller. This means the board is pretty unusable with UGS.

Make sure these things are working:
- [ ] Streaming file, pausing and stopping
- [ ]  Processing of Gcode states
- [ ]  Status report for machine coordinates
 - [ ] Status report for work coordinates
 - [ ] Jogging
 - [ ] Homing
 - [ ] Firmware settings
 - [ ] Overrides

justinclift, 2019-08-22:
If you hit blockers, let us know. :smile:

nobrand88, 2019-10-21:
> 
> ## Feature request
> 
> I've just got a TinyG board and hooked it up to UGS nightly. There where some basic stuff that's not working at the moment, like the initialization of the controller. This means the board is pretty unusable with UGS.
> 
> Make sure these things are working:
> 
>     * [ ]  Streaming file, pausing and stopping
> 
>     * [ ]   Processing of Gcode states
> 
>     * [ ]   Status report for machine coordinates
> 
>     * [ ]  Status report for work coordinates
> 
>     * [ ]  Jogging
> 
>     * [ ]  Homing
> 
>     * [ ]  Firmware settings
> 
>     * [ ]  Overrides
> 
> 
> Addresses issue #1111

Any updates to this issue? I have been struggling with bugginess of Chilipeppr (especially work-offsets), and Chilipeppr development seems to be abandoned on this forum. I would like to try UGS.

breiler, 2019-10-21:
I would like to have some other changes (#1260) in place before working on this.

nobrand88, 2019-10-21:
Ok, thanks. So it is not possible to get UGS functioning properly on tinyg as of now?

breiler, 2020-01-12:
I'm not able to test the homing feature on TinyG, is there anyone able to help me testing?

justinclift, 2020-01-12:
I can test with g2core later on today, if that's useful?

breiler, 2020-01-12:
I think it's already been verified with g2, but it would not hurt to test it again. 

I haven't been able to specifically test this with the tinyg firmware yet.

justinclift, 2020-01-12:
Ahhh, I only have a g2core setup here.

Hmm... @ril3y Do you still have a TinyG setup around?  UGS is looking to confirm that homing (still?) works ok. :smile:

ad0ndsm, 2020-11-22:
Homing is working on my TinyG with the current UGCS stable build with TinyG firmware 449.01. I do not have the ability to use the Jog panel, it is greyed out and unavailable.
The probing function seems to be functional as well.

ad0ndsm, 2020-11-22:
I was on the TinyG Edge firmware (449.01) I retrograded to the standard 440.20 and still do not have a jog panel in the stable build. In the nightly build (2.07 downloaded 11NOV2020) the Jog panel is working in both x32 and x64 versions of the platform.

ad0ndsm, 2020-11-23:
Apparently the firmware builds are reporting back to UGCS differently. I have verified they are configured via settings for JSON and such as near as I can tell. 
Both are set to Text mode, not JSON.
I have tried all 3 settings for verbosity, no jog on 449.
Going to look into 440 and probing, which is why I went to the edge version.

here is the console output between the two different firmware builds:

**** Connected to COM3 @ 115200 baud ****
[ready] {"r":{"fv":0.970,"fb":440.20,"hp":1,"hv":8,"id":"3X2204-PQ2","msg":"SYSTEM READY"},"f":[1,0,0,8881]}
>>> {ej:1}
>>> {sr:{posx:t, posy:t, posz:t, mpox:t, mpoy:t, mpoz:t, plan:t, vel:t, unit:t, stat:t, dist:t, frmo:t, coor:t}}
>>> {jv:4}
>>> {qv:0}
>>> {sv:1}
>>> {sr:n}
{"r":{"ej":1},"f":[1,0,7,2003]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"plan":0,"vel":0.00,"unit":1,"stat":1,"dist":0,"frmo":1,"coor":1}},"f":[1,0,109,1782]}
{"sr":{"plan":0,"vel":0.00,"unit":1,"dist":0,"frmo":1,"coor":1}}
{"r":{"jv":4},"f":[1,0,7,7333]}
{"r":{"qv":0},"f":[1,0,7,8353]}
{"r":{"sv":1},"f":[1,0,7,9250]}
{"r":{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"plan":0,"vel":0.00,"unit":1,"stat":1,"dist":0,"frmo":1,"coor":1}},"f":[1,0,7,3096]}
**** Connection closed ****

**** Connected to COM3 @ 115200 baud ****
{"sr":{"fv":0.970,"fb":449.01,"hp":1,"hv":0,"id":"3X2204-PQ2","msg":"Loading configs from EEPROM"}}
{"sr":{"fv":0.970,"fb":449.01,"hp":1,"hv":8,"id":"3X2204-PQ2","msg":"SYSTEM READY"}}