winder/Universal-G-Code-Sender #1263: Issue building ugs-pendant (same results as closed issue #1245)

rickromkey, 2019-08-22:
## Feature request

## Problem description 

When attempting to run on Raspberry Pi, failing when building ugs-pendant.

### Expected Behavior

Expect a completed build

### Actual Behavior

[INFO] ------------------------------------------------------------------------
[INFO] Building ugs-pendant 2.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) @ ugs-pendant ---
[INFO] Installing node version v10.15.0
[INFO] Downloading https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz to /root/.m2/repository/com/github/eirslett/node/10.15.0/node-10.15.0-linux-x86.tar.gz
[INFO] No proxies configured
[INFO] No proxy was configured, downloading directly
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  1.223 s]
[INFO] ugs-core ........................................... SUCCESS [ 10.816 s]
[INFO] ugs-pendant ........................................ FAILURE [  3.446 s]
[INFO] ugs-cli ............................................ SKIPPED
[INFO] ugs-classic ........................................ SKIPPED
[INFO] ugs-platform-parent ................................ SKIPPED
[INFO] ugs-platform-branding .............................. SKIPPED
[INFO] ugs-platform-ugslib ................................ SKIPPED
[INFO] ugs-platform-ugscore ............................... SKIPPED
[INFO] ugs-platform-visualizer ............................ SKIPPED
[INFO] ugs-platform-plugin-workflow ....................... SKIPPED
[INFO] ugs-platform-gcode-editor .......................... SKIPPED
[INFO] ugs-platform-surfacescanner ........................ SKIPPED
[INFO] ProbeModule ........................................ SKIPPED
[INFO] DowelModule ........................................ SKIPPED
[INFO] GcodeTools ......................................... SKIPPED
[INFO] ugs-platform-plugin-jog ............................ SKIPPED
[INFO] ugs-platform-plugin-setup-wizard ................... SKIPPED
[INFO] ugs-platform-welcome-page .......................... SKIPPED
[INFO] ugs-platform-app ................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 19.447 s
[INFO] Finished at: 2019-08-22T16:20:34+00:00
[INFO] Final Memory: 22M/52M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal com.github.eirslett:frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) on project ugs-pendant: Could not download Node.js: Got error code 404 from the server. -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :ugs-pendant

### Steps to Reproduce the Problem

  1. Virgin install of Pi.  Added Java, maven, etc.  Fails every time.
 

### Specifications
#### Version

Latest build.

#### Operating system

Linux on RaspberryPI
#### Platform

Pi 3b

#### Other

mvn -version
Apache Maven 3.3.9
Maven home: /usr/share/maven
Java version: 1.8.0_222, vendor: Raspbian
Java home: /usr/lib/jvm/java-8-openjdk-armhf/jre
Default locale: en_GB, platform encoding: UTF-8
OS name: "linux", version: "4.9.80-v7+", arch: "arm", family: "unix"

breiler, 2019-08-23:
I finally got my RPI up and running... 
It's running raspbian jessie

I'm running a later version of maven, but that shouldn't make any difference as deactivating profile has been around since Maven 2.0.10:
```
mvn -version
Apache Maven 3.6.1 (d66c9c0b3152b2e69ee9bac180bb8fcc8e6af555; 2019-04-04T19:00:29Z)
Maven home: /opt/apache-maven-3.6.1
Java version: 1.8.0_65, vendor: Oracle Corporation, runtime: /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre
Default locale: en_GB, platform encoding: UTF-8
OS name: "linux", version: "4.9.35-v7+", arch: "arm", family: "unix"
```

I run the build using this:
```
git clone https://github.com/winder/Universal-G-Code-Sender.git
cd  Universal-G-Code-Sender
mvn install -P'!create-pendant-ui-web'
```

And eventually got a successful build:
```
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0-SNAPSHOT:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  1.415 s]
[INFO] ugs-core ........................................... SUCCESS [01:15 min]
[INFO] ugs-pendant ........................................ SUCCESS [  4.394 s]
[INFO] ugs-cli ............................................ SUCCESS [ 19.749 s]
[INFO] ugs-classic ........................................ SUCCESS [ 16.922 s]
[INFO] ugs-platform-parent ................................ SUCCESS [  0.751 s]
[INFO] ugs-platform-branding .............................. SUCCESS [  6.366 s]
[INFO] ugs-platform-ugslib ................................ SUCCESS [ 24.540 s]
[INFO] ugs-platform-ugscore ............................... SUCCESS [  9.857 s]
[INFO] ugs-platform-visualizer ............................ SUCCESS [ 13.224 s]
[INFO] ugs-platform-plugin-workflow ....................... SUCCESS [  6.715 s]
[INFO] ugs-platform-gcode-editor .......................... SUCCESS [  7.224 s]
[INFO] ugs-platform-surfacescanner ........................ SUCCESS [  7.397 s]
[INFO] ProbeModule ........................................ SUCCESS [ 11.874 s]
[INFO] DowelModule ........................................ SUCCESS [ 35.595 s]
[INFO] GcodeTools ......................................... SUCCESS [  6.873 s]
[INFO] ugs-platform-plugin-jog ............................ SUCCESS [ 24.740 s]
[INFO] ugs-platform-plugin-setup-wizard ................... SUCCESS [  9.120 s]
[INFO] ugs-platform-welcome-page .......................... SUCCESS [  7.570 s]
[INFO] ugs-platform-app ................................... SUCCESS [ 40.965 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  05:34 min
[INFO] Finished at: 2019-08-23T13:10:14Z
[INFO] ------------------------------------------------------------------------
```

rickromkey, 2019-08-24:
I'm still failing.  Looking at the error, it's downloading  **https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz** which doesn't exist.  I manually downloaded the 64 bit version and created a sym link to it using the x86 name and got a couple steps further.  Then had to do similar when the image was extracted.  Eventually it all blew up again.

breiler, 2019-08-25:
What's the output from the command:
```git log --pretty=format:"%cn - %s - %h on %cd" | head -10```

breiler, 2019-08-29:
@rickromkey any updates?

cebess, 2019-10-10:
I have the same issue.
I am trying to install Universal-G-Code-Sender on a Pi 2.
It is running Linux raspberrypi 4.9.80-V7
My Node.js version is:
My Mavin version is: 3.6.2
Java: 1.8.0_65

This is the first time I've installed the software so I don't know what to expect.

When I tried to use the command you mentioned: mvn install -P'!create-pendant-ui-web'

I eventually saw:
[INFO] Installing node version v10.15.0
[INFO] Downloading https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-x86.tar.gz to /home/pi/.m2/repository/com/github/eirslett/node/10.15.0/node-10.15.0-linux-x86.tar.gz
[INFO] No proxies configured
[INFO] No proxy was configured, downloading directly
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0-SNAPSHOT:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  2.971 s]
[INFO] ugs-core ........................................... SUCCESS [02:51 min]
[INFO] ugs-pendant ........................................ FAILURE [  7.378 s]
[INFO] ugs-cli ............................................ SKIPPED
[INFO] ugs-classic ........................................ SKIPPED
[INFO] ugs-platform-parent ................................ SKIPPED
[INFO] ugs-platform-branding .............................. SKIPPED
[INFO] ugs-platform-ugslib ................................ SKIPPED
[INFO] ugs-platform-ugscore ............................... SKIPPED
[INFO] ugs-platform-visualizer ............................ SKIPPED
[INFO] ugs-platform-plugin-workflow ....................... SKIPPED
[INFO] ugs-platform-gcode-editor .......................... SKIPPED
[INFO] ugs-platform-surfacescanner ........................ SKIPPED
[INFO] ProbeModule ........................................ SKIPPED
[INFO] DowelModule ........................................ SKIPPED
[INFO] GcodeTools ......................................... SKIPPED
[INFO] ugs-platform-plugin-jog ............................ SKIPPED
[INFO] ugs-platform-plugin-setup-wizard ................... SKIPPED
[INFO] ugs-platform-welcome-page .......................... SKIPPED
[INFO] ugs-platform-app ................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  03:11 min
[INFO] Finished at: 2019-10-10T13:14:17-04:00
[INFO] ------------------------------------------------------------------------
[WARNING] The requested profile "| create-pendant ui-web" could not be activated because it does not exist.
[ERROR] Failed to execute goal com.github.eirslett:frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) on project ugs-pendant: Could not download Node.js: Got error code 404 from the server. -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :ugs-pendant

Any ideas on what to do?

breiler, 2019-10-10:
@cebess I didn't get any response from @rickromkey.

What's the output if you run this from the source directory: `git log --pretty=format:"%cn - %s - %h on %cd" | head -10`

cebess, 2019-10-10:
BY the way the output from:
git log --pretty=format:"%cn - %s - %h on %cd" | head -10

was:
GitHub - Merge pull request #1280 from breiler/feature/gcodeeditor - cacd709c on Tue Oct 8 20:13:56 2019 +0200
Joacim Breiler - Fixed bug with npe:s when no movement was defined. - 7a360643 on Tue Oct 8 20:02:42 2019 +0200
Joacim Breiler - Moved back the menu item for showing the editor. Fixed an issue where the wrong command was highlighted in the visualizer - 965c9b26 on Tue Oct 8 18:53:05 2019 +0200
Joacim Breiler - Fixed problem with lines highlighter not working in the editor. - c25e6d8e on Tue Oct 8 15:43:58 2019 +0200
Joacim Breiler - Some cleanups after code review - f4c00349 on Mon Oct 7 20:28:03 2019 +0200
Joacim Breiler - Fixed version number - 6b30a11c on Sun Oct 6 20:48:05 2019 +0200
Joacim Breiler - Fixed tests - 6eceef9e on Sun Oct 6 20:38:42 2019 +0200
Joacim Breiler - Added some javadoc - fd4b276b on Sun Oct 6 20:31:41 2019 +0200
Joacim Breiler - Added syntax highlighter - 45ad03c4 on Sun Oct 6 20:23:20 2019 +0200
Joacim Breiler - Uses default colors

cebess, 2019-10-10:
pi@raspberrypi:~/Universal-G-Code-sender $ git log --pretty=format:"%cn - %s - %h on %cd" | head -10

 

GitHub - Merge pull request #1280 from breiler/feature/gcodeeditor - cacd709c on Tue Oct 8 20:13:56 2019 +0200

Joacim Breiler - Fixed bug with npe:s when no movement was defined. - 7a360643 on Tue Oct 8 20:02:42 2019 +0200

Joacim Breiler - Moved back the menu item for showing the editor. Fixed an issue where the wrong command was highlighted in the visualizer - 965c9b26 on Tue Oct 8 18:53:05 2019 +0200

Joacim Breiler - Fixed problem with lines highlighter not working in the editor. - c25e6d8e on Tue Oct 8 15:43:58 2019 +0200

Joacim Breiler - Some cleanups after code review - f4c00349 on Mon Oct 7 20:28:03 2019 +0200

Joacim Breiler - Fixed version number - 6b30a11c on Sun Oct 6 20:48:05 2019 +0200

Joacim Breiler - Fixed tests - 6eceef9e on Sun Oct 6 20:38:42 2019 +0200

Joacim Breiler - Added some javadoc - fd4b276b on Sun Oct 6 20:31:41 2019 +0200

Joacim Breiler - Added syntax highlighter - 45ad03c4 on Sun Oct 6 20:23:20 2019 +0200

Joacim Breiler - Uses default colors, removed unused code. - 77caaaa3 on Sat Oct 5 20:16:33 2019 +0200

 

---

Charles Bess

breiler, 2019-10-10:
Sorry, I can't reproduce this error. I think your best option is to either download a build from https://github.com/winder/Universal-G-Code-Sender#downloads 

Or to build it on another computer and upload the `ugs-platform/application/target/ugs-platform-app-2.0-SNAPSHOT.zip`or `ugs-classic/target/UniversalGcodeSender.jar` to the Pi.

cebess, 2019-10-10:
I believe that is where I got it from using the command:
git clone https://github.com/winder/Universal-G-Code-sender

I'll try and download it from a browser?!?

breiler, 2019-10-10:
You don't need to download the code and build it yourself, just download one of the prebuilt packages, unzip and run.

You'll find some instructions here: https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi

cebess, 2019-10-10:
Thanks for your assistance. I’ll let you know and do a blog post about the process, once I have it all working.

 

 

---

Charles Bess

closed 2019-10-26 by breiler

DanRegalia, 2020-05-04:
@breiler Quick question, I'm a little new to running this on a pi.  How do you run the platform and skip errors, etc for the first run.  I know everything is installed, but I'm using a different version of node/npm. Thanks for your help.

breiler, 2020-05-04:
@DanRegalia what are you trying to do?

Download the latest version, unpack and run:
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#download-and-start-ugs-platform

Or build it from scratch, this does however require you to inactivate the web pendant build as the build step for this doesn't work on ARM-processors:
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi#download-the-source-and-build

janlinders, 2020-12-21:
I have come across a similar issue now on Linux Ubuntu.
Not able to compile the GIT version.
See logging below.

I skipped the 'pendant' compile since this runs into a SEGMENTATION FAULT.
After compile here it does start the application with the startup image but after that it closes.

`INFO] ------------------< com.willwinder:ugs-platform-app >-------------------
[INFO] Building ugs-platform-app 2.0.7-SNAPSHOT
[INFO] --------------------------[ nbm-application ]---------------------------
[INFO] 
[INFO] --- nbm-maven-plugin:4.4:run-platform (default-cli) @ ugs-platform-app ---
[INFO] Executing: /bin/sh -c '/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/bin/ugsplatform' '--userdir' '/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/userdir' '-J-Dnetbeans.logger.console=true' '-J-ea' '--branding' 'ugsplatform'
[INFO] OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
[INFO] WARNING: An illegal reflective access operation has occurred
[INFO] WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
[INFO] WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
[INFO] WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
[INFO] WARNING: All illegal access operations will be denied in a future release
[INFO] -------------------------------------------------------------------------------
[INFO] >Log Session: Monday, December 21, 2020 at 1:23:38 PM Central European Standard Time
[INFO] >System Info: 
[INFO]   Product Version         = Universal Gcode Platform 12.0-631bd69cd6112b1cc4c892c24e3e605b1ba04241
[INFO]   Operating System        = Linux version 5.8.0-33-generic running on amd64
[INFO]   Java; VM; Vendor        = 13.0.4; OpenJDK 64-Bit Server VM 13.0.4+8-Ubuntu-1; Private Build
[INFO]   Runtime                 = OpenJDK Runtime Environment 13.0.4+8-Ubuntu-1
[INFO]   Java Home               = /usr/lib/jvm/java-13-openjdk-amd64
[INFO]   System Locale; Encoding = en_US (ugsplatform); UTF-8
[INFO]   Home Directory          = /home/admin
[INFO]   Current Directory       = /home/admin/Universal-G-Code-Sender
[INFO]   User Directory          = /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/userdir
[INFO]   Cache Directory         = /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/userdir/var/cache
[INFO]   Installation            = /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/extra
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ide
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/etc
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ugsplatform
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/bin
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/java
[INFO]                             /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform
[INFO]   Boot & Ext. Classpath   = 
[INFO]   Application Classpath   = /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/boot.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-modules.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util-lookup.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util-ui.jar
[INFO]   Startup Classpath       = /home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-7.2.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-netbeans-libs-asm.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/core.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-openide-filesystems.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/core-base.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-commons-7.2.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-tree-7.2.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-openide-filesystems-compat8.jar:/home/admin/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ugsplatform/core/locale/core_ugsplatform.jar
[INFO] -------------------------------------------------------------------------------
[... cut ...]
[INFO] Inconsistency detected by ld.so: dl-lookup.c: 105: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  16.662 s
[INFO] Finished at: 2020-12-21T13:23:50+01:00
[INFO] ------------------------------------------------------------------------
`

janlinders, 2020-12-21:
SOme extra info on the Pendant failure.

INFO] --- frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) @ ugs-pendant ---
[INFO] Node v10.15.0 is already installed.
[INFO] NPM 6.4.1 is already installed.
[INFO] 
[INFO] --- frontend-maven-plugin:1.6:npm (npm install) @ ugs-pendant ---
[INFO] Running 'npm install' in /home/admin/Universal-G-Code-Sender/ugs-pendant/src/main/webapp
^[[B[INFO] 
[INFO] > core-js@3.6.4 postinstall /home/admin/Universal-G-Code-Sender/ugs-pendant/src/main/webapp/node_modules/@angular-devkit/build-angular/node_modules/core-js
[INFO] > node -e "try{require('./postinstall')}catch(e){}"
[INFO] 
[ERROR] Segmentation fault (core dumped)
[WARNING] npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/chokidar/node_modules/fsevents):
[WARNING] npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
[WARNING] npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.7 (node_modules/fsevents):
[WARNING] npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.7: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
[WARNING] npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/karma/node_modules/fsevents):
[WARNING] npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
[ERROR] 
[ERROR] npm ERR! code ELIFECYCLE
[ERROR] npm ERR! errno 139
[ERROR] npm ERR! core-js@3.6.4 postinstall: `node -e "try{require('./postinstall')}catch(e){}"`
[ERROR] npm ERR! Exit status 139
[ERROR] npm ERR! 
[ERROR] npm ERR! Failed at the core-js@3.6.4 postinstall script.
[ERROR] npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
[ERROR] 
[ERROR] npm ERR! A complete log of this run can be found in:
[ERROR] npm ERR!     /home/admin/.npm/_logs/2020-12-21T12_37_43_847Z-debug.log
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for ugs-parent 2.0.7-SNAPSHOT:
[INFO] 
[INFO] ugs-parent ......................................... SUCCESS [  0.598 s]
[INFO] ugs-core ........................................... SUCCESS [  3.092 s]
[INFO] ugs-pendant ........................................ FAILURE [ 12.407 s]
[INFO] ugs-cli ............................................ SKIPPED
[INFO] ugs-classic ........................................ SKIPPED

lamikr, 2021-04-27:
Does it help if you update the npm and nodeversion in ugs-pedant/pom.xml to version that you have installed locally? I had also problems until I updated my versions with this patch:

PR #1294 (2019-10-26): Fixed issue where application can't be compiled jdk11.
Upgraded libraries that isn't compatible with jdk11.
Fixes #1263 have tested on a clean rpi-installation using the instructions under https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi