winder/Universal-G-Code-Sender #1266: g2core: JsonNull on "Soft Reset"

norru, 2019-09-01:
Firmware: g2core 101.03 Edge
Controller: Arduino Due
Shield: custom

Product Version: Universal Gcode Platform 20190827
Java: 1.8.0_212; OpenJDK 64-Bit Server VM 25.212-b03
Runtime: OpenJDK Runtime Environment 1.8.0_212-8u212-b03-2-b03
System: Linux version 4.19.0-5-amd64 running on amd64; UTF-8; en_US (ugsplatform)
User directory: /home/nico/.ugsplatform/2.0-SNAPSHOT/dev
Cache directory: /home/nico/.ugsplatform/2.0-SNAPSHOT/dev/var/cache

OS: Debian/Devuan Ceres

To reproduce
- Start UGS Platform
- Connect to controller on port /dev/ttyACM0
- click on "Soft Reset"

Results:
- the exception below is reported
- connection and terminal become unresponsive until disconnected

Expected result:
- soft reset action is performed
- connection doesn't hang

```
java.lang.UnsupportedOperationException: JsonNull
	at com.google.gson.JsonElement.getAsString(JsonElement.java:191)
	at com.willwinder.universalgcodesender.TinyGUtils.hasNumericField(TinyGUtils.java:393)
	at com.willwinder.universalgcodesender.TinyGUtils.updateControllerStatus(TinyGUtils.java:197)
	at com.willwinder.universalgcodesender.TinyGController.updateControllerStatus(TinyGController.java:241)
	at com.willwinder.universalgcodesender.TinyGController.rawResponseHandler(TinyGController.java:211)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:851)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:222)
[catch] at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:244)
	at java.lang.Thread.run(Thread.java:748)
```

breiler, 2019-09-01:
The version 101.03 of g2core had a bug that made the board unresponsive waiting for it to become idle:
https://github.com/synthetos/g2/issues/367

I've never got the error message you've got, if you've got the opportunity, please try version: g2core 100.26.

norru, 2019-09-01:
Can't use 100.26. I don't remember which feature it was missing but it was only available lin latest Edge.

breiler, 2019-09-01:
That's unfortunate, the g2core bug has been fixed but is still waiting a new build.

norru, 2019-09-01:
I can build from source no problem!

breiler, 2019-10-30:
I've installed and tested with the latest g2core source and got the same error. The applied patch should fix that problem.

closed 2019-10-30 by breiler