winder/Universal-G-Code-Sender #1269: NPE @ GRBLUtils.java:471

maatinh, 2019-09-11:
I just used UGS for the first time with an Elekslaser-A3. It's freezing regularly :(
When started from the console I can see a NPE at GRBLUtils.java:471.
Looking at the code shows that it is possible that the String returned from the getStateFromStatusString()-method is null but this is not checked within getControllerStateFromStateString().

breiler, 2019-09-11:
What version of UGS are you running?
What version of GRBL firmware are you running?

Could you provide a dump of your GRBL settings ($$)

maatinh, 2019-09-11:
I downloaded the nighlty versions of UGS few hours ago. I'm running classic because the "Platform" crashes immediately on startup (running on latest Linux Mint). And btw classic crashes when trying to use the visualize feature.
GRBL is 0.9. I'll send the settings dump later today as the laser is currently running.

breiler, 2019-09-11:
Regarding the "Platform" crashing I'm really interested in your logs and the console message. They should be located in:
```
~/.ugsplatform/2.0-SNAPSHOT/dev/var/log/messages.log
```

I tried starting it on Ubuntu Buster with OpenJDK 12 and I also got i to crash with the same error as in issue as #1228. To solve that, try downgrading to java 8 and update the default java version:
```
sudo apt install openjdk-8-jdk
sudo update-alternatives --config java
sudo update-alternatives --config javac
```

The GRBL-version you are using is quite old (20150625). I need to download it and test it my self.
In the meantime you could try activating verbose logging and paste a copy of a status report line.

maatinh, 2019-09-11:
[messages.log.1.txt](https://github.com/winder/Universal-G-Code-Sender/files/3601252/messages.log.1.txt)

maatinh, 2019-09-11:
Now I downgraded to jdk8 and used the Platform-version. It didn't crash on startup but it stopped during engraving.
The message in the UI is:
ok
>>> X37.2Y56.4S208
ok
ok
>>> X36.9Y56.7S221
>>> X36.6Y57S202
[Error] An error was detected while sending 'G0X38.7Y54.9S0': error: Bad number format. Streaming has been paused.
**** The communicator has been paused ****

**** Pausing file transfer. ****

Unfortunately the Laser is not switched off when pausing. Here is the messages.log of this run:
[messages.log.txt](https://github.com/winder/Universal-G-Code-Sender/files/3601638/messages.log.txt)

PR #1342 (2020-02-25): Fixes problem where position is reported with an even number.
Fixes #1269 

Made the decimal point optional which now makes it possible to parse positions reported like this:
<Run,MPos:50.400,43.200,0.000,WPos:50000,42.600,0.000>
closed 2020-02-25 by winder