winder/Universal-G-Code-Sender #1270: File fails to visualize correctly

khturnings, 2019-09-14:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

DanielWeigl, 2019-09-15:
Can you try a more recent version, http://winder.github.io/ugs_website/download/#nightly-builds

This was most likely fixed here https://github.com/winder/Universal-G-Code-Sender/pull/1220

khturnings, 2019-09-15:
Tried with nightly build - August 27 2019 with same result.
[image]

khturnings, 2019-09-15:
I reprocessed the file in Fusion 360 and got the correct output this time.

Please put this down to human error. 

I will close this issue. Thank for your help

closed 2019-09-15 by khturnings

DanielWeigl, 2019-09-17:
It also looks strange in ncviewer.com:
[image]

Seems you have used a wrong postprocessor in fusion360, ive sofar not seen "inline" comments, like this:
```
G03 ( Circular CCW ) X0002.0550 Y0010.0000 I-0001.2500 J0000.0000 S90 ( Set laser power )
G03 ( Circular CCW ) X0004.5550 Y0010.0000 I0001.2500 J0000.0000 S90 ( Set laser power )
```

(the ` ( Circular CCW )`)