winder/Universal-G-Code-Sender #1271: Tiler not working

jappiemike, 2019-09-17:
I have tried all the most recent versions of UGS and cannot get the tiler to work, it always comes up with rapids lines to where the objects should be but no object, It also shows the program running as the dowel_program and a long line of numbers. There is no tiler program in the folder, I'm using a mac running 10.13.6 with java 8 something.

breiler, 2019-10-26:
It works for me, do you mind sharing your gcode program that we can try?

[image]

The filename dowel_program is just a generated for the temporary file. It should of course be something like tiler_program so that's an improvement that can be made.

jappiemike, 2019-10-26:
Hi and thanks for taking the time to help. I just used this file which is the cutout of some perspex circles and was generated by carbide create..

and this is the output after the tiler

it shows the path between the shapes but no code for the actual circles. I’ve tried many of the older versions of UGS but haven’t found one that works.
Cheers
Mike

breiler, 2019-10-28:
It seems that it has problems parsing some of the comments in the file. As a workaround, remove the lines with parenthesis.

jappiemike, 2019-11-03:
That worked, I was hoping to just comment out the parenthicated (?) lines but I had to remove them which is a bummer as they have information about the stock size and tool. Thanks for the help.
Cheers
Mike

breiler, 2020-04-10:
This has been fixed in #1364

closed 2020-04-10 by breiler