winder/Universal-G-Code-Sender #1272: This G-Code works with bCNC but halts with UGS

CliveMcCarthy, 2019-09-18:
I ran this test code on the same rig, same GRBL, same computer with bCNC. UGS halts after about 45 minutes. There is no error message.

[hello_image.zip](https://github.com/winder/Universal-G-Code-Sender/files/3627388/hello_image.zip)

breiler, 2019-09-18:
Classic or Platform edition?
Which version of UGS are you running?
What operating system are you using?

CliveMcCarthy, 2019-09-18:
UGS
        Platform    20190814
        Java                   1.8.0_222
        Linux                  4.15.0-62
      
    On 9/18/19 9:31 AM, Joacim Breiler
      wrote:
    
    
      
      Classic or Platform edition?
        Which version of UGS are you running?
        What operating system are you using?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/winder/Universal-G-Code-Sender/issues/1272?email_source=notifications\u0026email_token=ACPQYUPDEWBF4QLTL6JUMWDQKJJXDA5CNFSM4IYA52FKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7AVIIY#issuecomment-532763683",
"url": "https://github.com/winder/Universal-G-Code-Sender/issues/1272?email_source=notifications\u0026email_token=ACPQYUPDEWBF4QLTL6JUMWDQKJJXDA5CNFSM4IYA52FKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7AVIIY#issuecomment-532763683",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

breiler, 2019-10-28:
Sorry for the late reply.

What processor architecture are you running?
`uname -a`

How much memory is available on the system?

I've just run the whole file without any problems.

breiler, 2019-12-12:
I suspect that this is due to running jSerialComm driver on Linux, but without further answers from user it's hard to solve. Closing the issue.

closed 2019-12-12 by breiler

jes, 2020-11-11:
If the JSerialComm driver doesn't work properly on Linux, should we consider making JSSC the default?