winder/Universal-G-Code-Sender #1274: Nightly Platform (09/24/19) on Opensuse problem

falviani, 2019-09-25:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

Running OpenSuse Tumbleweed. When I open the platform nightly it gets to the point of opening an empty window titled 'Nightly Build' then quits immediately. This has the standard linux Java installed, rather than the Oracle flavor.

breiler, 2019-09-25:
This sound like a duplicate of #1228. Are you running Java 8?

Type this in a console:
`java -version`

falviani, 2019-09-25:
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (IcedTea 3.13.0) (build 1.8.0_222-b10
suse-1.1-x86_64)
OpenJDK 64-Bit Server VM (build 25.222-b10, mixed mode)

I have nothing equivalent to the Ubuntu alternatives that I know of.

Hope this is useful.

falviani, 2019-09-25:
As a test, I downloaded the classic nightly, and it opened fine.

falviani, 2019-09-29:
Also, there is no ~/ugs folder in Linux. Perhaps there's a permissions issue, but the ownership for the binary is shown as frank/users.

falviani, 2019-10-23:
As a test, I manually created the ~/ugs folder and tried again, No change.

winder, 2019-10-23:
It's a hidden directory, so you'll need to look at ~/.ugs

falviani, 2019-10-23:
Changed the name (I overlooked the '.' earlier) and just in case made sure
the permissions were OK. Unfortunately, no change - still aborts when
bringing up the 'nightly build' warning window.

falviani, 2019-11-16:
Tried the nightly on 11-19-19. No change. still closes automatically.

closed 2019-11-16 by falviani