winder/Universal-G-Code-Sender #1275: Issue using UGS in asus tinker board

locoxhun, 2019-09-25:
Hello everyone, 

I'm trying to use UGS in my Asus tinker board but i just can't. I've tried A LOT. 

Whenever i try to open the connection (using grbl 0.9j), here's what i got:

Grbl 0.9j ['$' for help]
**** Connected to ttyUSB0 @ 115200 baud ****
>>> $$
Grbl 0.9j ['$' for help]
>>> $G
>>> $$
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
$0 = 10    (step pubool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, b0 (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing0.000 (x, step/mm)
$101 = 250.000    (y, step/mm)
$102 = 250.000    (z,  mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)

And it stops there, i cant move the axis, and the firmware settins are incomplete.

I've also tried to use JSSC driver but nothing changed.

I'm using the official TinkerOs 2.0.11.

Java 8 installed. Running UGS using start.sh.
Please, i need help with this, i've been suffering for days.

breiler, 2019-09-26:
Which version of UGS are you running?
The GRBL-firmware is pretty old (20163016) is it possible to upgrade?
What kind of Arduino are you using, cheap China knock off or a genuine Arduino?
Do you have another computer to try with?
Do you have another USB cable to try with, could be signal interference?

locoxhun, 2019-09-26:
Hello Mr. Breiler,

First of all thank you for the replay, you are the first to help me in days.
And sorry for the poor information i gave.
Ok, so answering your questions:

-I'm running the latest version of UGS from the official Github page (2.0 nightly) i tried both classic and platform version;
-I do not know if it is possible to upgrade the GRBL. I'm currently using GRBL 0.9j;
-I'm using (unfortunately) the cheap China one, i do have a genunine one, but it's ATmega is burnt;
-Yes i do. And it works perfectly. I do have a notebook with Windows 10, and the UGS connects to the same arduino and it runs without any problem.
-Yes, i've tried three different cables, all three works whenever i try on my notebook, but not in the Tinker Board.

When connecting the Arduino to the Tinker Board, lsusb (listing usb) shows this:

linaro@tinkerboard:~$ lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:481a Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

There it is the Arduino ch340 usb adapter:
"Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter"

And there's something odd that happens:
When i upload GRBL 0.8j that uses 9600 baud rate, it "connects". I receive the "ok" message in the end, i'm able to move manualy the axis, but the firmware is also not complete.

Here, it follows:

**** Connected to ttyUSB0 @ 9600 baud ****
Grbl 0.8c ['$' for help]
>>> $$
Grbl 0.8c ['$' for help]
>>> $G
>>> $$
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F250.000]
ok
$0 = 250.000    (x, step/mm)
$1 = 250.000    (y, step/mm)
$2 = 250.000    (z, step/mm)
$3 = 10    (step pulse, usec)
$4 = 250.000    (default feed, mm/min)
$5 = 500.000    (default seek, mm/min)
$6 = 192    (step port invert mask, int:11000000)
$7 = 25    (step idle delay, msec)
$8 = 10.000    (acceleration, mm/sec^2)
$9 = 0.050    (junction deviation, mm)
$10 = 0.100    (arc, mm/segment)
$11 = 25    (n-arc correction, int)
$12 = 3    (n-decimals, int)
$13 = 0    (report inches, bool)
$14 = 1    (auto start, bool)
$15 = 0    (invert step enable, bool)
$16 = 0    (hard limits, bool)
$17 = 0    (homing cycle, bool)
$18 = 0    (homing dir invert mask, int:00000000)
$19 = 25.000    (homing feed, mm/min)
$20 = 250.000    (homing seek, mm/min)
$21 = 100    (homing debounce, msec)
$22 = 1.000    (homing pull-off, mm)
ok
>>> $G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F250.000]
ok

And that is how far i got, and can not move foward from this :(

Also it might be helpful to know that, whenever i try to upload grbl or any sketch using arduino IDE in the Tinker Board, it gives the following message:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avr_read(): error reading address 0x0000
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2
avrdude: stk500_recv(): programmer is not responding

BUT, awkwardly it sends the sketch anyways. How do i know that? When i send a blink, the led blinks; when i send grbl, it successfuly connects to UGS in my notebook. At least i think it is sending the sketchs. Maybe it is sending but it wont work properly.

Don't know if it is related to the main issue. But it is worth mentioning.

Anyways, i hope this information can help you help me hahaha!

Thank you!!

locoxhun, 2019-09-26:
Hey there!

Right know i'm using Armbian in another SD card, and with it i managed to connect grbl 0.9j, with 115200 of baud rate, but still, it does not connects to the complet firmware.

It should look like:

Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, step/mm)
$102 = 250.000    (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)
ok
>>> $G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

but it is showing:

Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
=0 (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviatiance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool$22=0 (homing cycle, bool)
$23 = 0    (homing dir invert mask:000000 mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debng pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, stmm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max ratex rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.200.000    (z max travel, mm)
ok
>>> $G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

As you can see, it has a few lines missing.