winder/Universal-G-Code-Sender #1276: GCODE not compatible with grbl  1.1  HELP!

oraherb, 2019-10-03:
please help me im a new to cnc  so i got this problem with gcode when i try to send this gcode to universal gcode sender..it gives me an error    "an error was detected while sending  OS30: error:1"

the gcode i was sending is like this

G21 (metric ftw)
G90 (absolute mode)
G92 X0.00 Y0.00 Z0.00 (you are here)

M300 S30 (pen down)
G4 P50 (wait 50ms)
M300 S50 (pen up)
G4 P50 (wait 50ms)
M18 (disengage drives)
M01 (Was registration test successful?)
M17 (engage drives if YES, and continue)

M01 (Plotting layer 'Calque 1')
(Polyline consisting of 63 segments.)
G1 X13.99 Y0.73 F3000.00
M300 S30.00 (pen down)
G4 P50 (wait 50ms)
G1 X11.60 Y1.51 F3000.00
G1 X10.51 Y1.89 F3000.00
G1 X9.97 Y2.14 F3000.00
G1 X9.43 Y2.38 F3000.00
G1 X8.54 Y2.92 F3000.00
G1 X4.32 Y6.01 F3000.00
G1 X2.84 Y7.59 F3000.00
G1 X1.72 Y9.24 F3000.00
G1 X1.50 Y9.61 F3000.00
G1 X1.44 Y9.88 F3000.00
G1 X1.37 Y10.14 F3000.00
G1 X1.37 Y10.44 F3000.00
G1 X1.37 Y10.75 F3000.00
G1 X1.26 Y10.82 F3000.00
G1 X0.80 Y11.01 F3000.00
G1 X0.46 Y11.14 F3000.00
G1 X0.52 Y11.16 F3000.00
G1 X0.57 Y11.19 F3000.00
G1 X0.91 Y11.07 F3000.00
G1 X1.32 Y10.89 F3000.00
G1 X1.39 Y10.83 F3000.00
G1 X1.66 Y11.17 F3000.00
G1 X1.94 Y11.51 F3000.00
G1 X2.18 Y11.64 F3000.00
G1 X3.64 Y12.19 F3000.00

(Polyline consisting of 63 segments.)
G1 X12.29 Y37.91 F3000.00
M300 S30.00 (pen down)
G4 P50 (wait 50ms)
G1 X12.59 Y37.96 F3000.00
G1 X12.74 Y37.96 F3000.00
G1 X12.54 Y37.91 F3000.00
G1 X12.24 Y37.87 F3000.00
G1 X12.14 Y37.87 F3000.00
G1 X12.29 Y37.92 F3000.00
G1 X12.29 Y37.91 F3000.00
M300 S50.00 (pen up)
G4 P50 (wait 50ms)

(Polyline consisting of 63 segments.)
G1 X12.85 Y38.02 F3000.00
M300 S30.00 (pen down)
G4 P50 (wait 50ms)
G1 X13.07 Y38.02 F3000.00
G1 X13.18 Y37.99 F3000.00
G1 X12.96 Y37.99 F3000.00
G1 X12.74 Y37.99 F3000.00
G1 X12.85 Y38.02 F3000.00
M300 S50.00 (pen up)
G4 P50 (wait 50ms)

(end of print job)
M300 S50.00 (pen up)
G4 P50 (wait 50ms)
M300 S255 (turn off servo)
G1 X0 Y0 F3000.00
G1 Z0.00 F150.00 (go up to finished level)
G1 X0.00 Y0.00 F3000.00 (go home)
M18 (drives off)

breiler, 2019-10-03:
The M300 looks suspicious along with some other M-commands... Those commands are not supported by GRBL. How did you generate this gcode?

oraherb, 2019-10-03:
@breiler  i create gcode in inkscape and it is automaticaly created

breiler, 2019-10-03:
Which plugin?

oraherb, 2019-10-03:
@breiler but when i download a sample gcode  

G17
G21
G90
G0 Z5.0000
G0 X0.0000 Y0.0000 S0 M3
G0 X36.9749 Y39.5443 Z0.0250
G1   Z-0.1000 F360.0
G1 X11.8301   F1200.0
G1  Y8.8481 
G1 X36.9749  
G1  Y39.5443 
G0   Z0.0250
G0 X22.9330 Y68.2811 
G1   Z-0.1000 F360.0
G3 X175.4342 Y70.2404 I75.6307 J49.2305 F1200.0
G0   Z0.0250
G0 X158.7799 Y37.5849 
G1   Z-0.1000 F360.0
G1  Y7.8684  F1200.0
G1 X184.5778  
G1  Y37.5849 
G1 X158.7799  
G0   Z0.0250
G0 X98.6938 Y127.3876 
G1   Z-0.1000 F360.0
G1 X107.3871 Y129.7169  F1200.0
G1 X110.8871 Y123.6548 
G1 X106.3871 Y115.8605 
G1 X97.3871  
G1 X88.2440 Y127.3876 
G1 X97.0610 Y135.2249 
G1 X110.0610  
G1 X119.9199 Y127.3876 
G1 X117.9605 Y113.0191 
G1 X109.4701 Y103.2225 
G1 X94.4701  
G1 X79.1005 Y113.6722 
G1 X77.7943 Y131.6328 
G1 X85.7943 Y145.4892 
G1 X98.3513 Y148.8538 
G1 X114.3513  
G1 X127.4306 Y142.4091 
G1 X136.5742 Y129.0203 
G1  Y112.0203  

its works but im wondering how can i make a gcode like that  and  i need the servo for pen up and down

oraherb, 2019-10-03:
@breiler sorry im new to this what im still learning  what plugin??

oraherb, 2019-10-03:
please help guys

breiler, 2019-10-03:
Dude it's impossible to help you with this little information... Did you follow a guide?

Let's start with the hardware and GRBL-firmware:
* GRBL does not support servos so I suspect that you are not using a standard build (https://github.com/gnea/grbl). Where did you download it?
* How are you connecting the servo to the Arduino for the pen?

The plugin for Inkscape is quite important and is really dependant on the right hardware.
I couldn't find anything on "MI Inkscape plugin" or "MI GRBL Z-AXIS SERVO Controller".

It looks like the Gcode you posted first comes from the "Unicorn plugin" (https://github.com/martymcguire/inkscape-unicorn) which requires the right hardware. More specifically it was designed for "MakerBot CupCake CNC" which I believe uses Marlin firmware popular with 3D-printers. 

Ex. the M300-command is used for playing a tone using a specific frequency which I guess can be used for signalling a servo. (http://marlinfw.org/docs/gcode/M300.html)

oraherb, 2019-10-03:
@breiler im using this grbl  (https://github.com/gnea/grbl) and  its not modified and using ardruino nano   with  stepstick drv8825   i connect the servo with  DIGITAL PIN 11  and GND  and 5V on arduino

oraherb, 2019-10-04:
@breiler  but my issue is not the pen and servo my gcode that i send from inkscape gives me an error so how can i create a gcode that can run in UNIVERSAL GCODE SENDER please help to make a gcode that compatible with (https://github.com/gnea/grbl) this grbl thats only my problem how to make compatible gcode sir please thanks

breiler, 2019-10-04:
There are alot of options, when I started I used Easel from Iventables:
http://easel.inventables.com/

It can export gcode, have look at the following link for instructions:
https://inventables.zendesk.com/hc/en-us/articles/360012762533-Walkthrough-Tutorial-Offline-Easel-Usage

Other than that you can have a look at this page from Shapeoko and pick a tool that suits your needs:
https://wiki.shapeoko.com/index.php/CAM

whitetd, 2019-10-04:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.instructables.com/id/How-to-Control-a-Servo-Using-GRBL/&ved=2ahUKEwjKwqHg_oHlAhWCFjQIHbGtDR4QFigAMAB6BAgHEAE&usg=AOvVaw1GGGoz6Q3cgRSNr3JBIKxK&cshid=1570171014511

oraherb, 2019-10-04:
@breiler so do i have to pay to download the easel?

breiler, 2019-10-04:
No, it's a web based software and requires you to create an account at inventables.

oraherb, 2019-10-04:
@breiler thanks it work bro

closed 2019-10-04 by breiler