winder/Universal-G-Code-Sender #1279: help please how can i configure gbrl to have a pen up and down for my pen cnc

oraherb, 2019-10-04:
hi guys can someone help me how to configure this grbl (https://github.com/bdring/Grbl_Pen_Servo)  to work with my servo for up and down movement? im using windows 10 and  ugs 
my hardware is arduino nano   and i connect the orange wire or pulse  of the servo to  pin D11  on arduino    how can i make it work to have a pen type cnc
i follow all the instruction from here (https://www.instructables.com/id/How-to-Control-a-Servo-Using-GRBL/)but nothing work

svenhb, 2019-10-05:
I used https://github.com/cprezzi/grbl-servo
Just send Sx for spindle PWM to control a servo - x between 0 and 1000
Short description in german for an egg-bot with servo (but with link to google translator): http://svenhb.bplaced.net/?CNC___Eggbot___Eggbot_via_GRBL-Plotter

oraherb, 2019-10-06:
@svenhb  bro so i have to change the gcode  to trigger the servo for pen up and down? what will i insert to the gcode?

svenhb, 2019-10-06:
Yes, for example send "M3 S50" for pen down and "M3 S500" for pen up (or M4).
The correct values for S you need to figure out.
If the servo need some time for motion you may add the command "G4 P1" after pen down/up command for 1 second delay

oraherb, 2019-10-06:
@svenh  bro thanks but how can i figure out the correct value for my S?? and it gives me an error so whats wrong i follow the instruction in your website please help me bro

oraherb, 2019-10-06:
[image]
heres my error   bro  @svenhb

svenhb, 2019-10-06:
You send M17 and M18, this commands are not known by grbl, and also not known for other CNC - check known M-Codes here: http://linuxcnc.org/docs/html/gcode/m-code.html

svenhb, 2019-10-06:
How do you generate your GCode?

oraherb, 2019-10-06:
@svenhb  in inkscape and it has an unicorn plugin bro

svenhb, 2019-10-06:
You may save your inkscape path and open it in GRBL-Plotter... 
But here we are in the Universal-G-Code-Sender area...

oraherb, 2019-10-06:
@sve
[image]   how can i generate gcode for my pen plotter bro? 
nhb

svenhb, 2019-10-06:
Drag your SVG and drop it on the 2D view
In Setup section you define what to do at pen up/down action

oraherb, 2019-10-06:
@svenhb  ok i will try it thanks hope it works just a sec...

oraherb, 2019-10-06:
@svenhb bro your awesome  but  i have a problem  where should i define the pen up and down action i cannot find  it where to write the  code for  pen example (m3 s50
[image]
)

oraherb, 2019-10-06:
@svenhb  bro it means a lot to me when i make this project works im a new to this still learning

svenhb, 2019-10-06:
In setup - graphics import - GCode generation, on the far right, enable Servo control.
Via Main GUI "Control/Jogging" you need to switch off / on the spindel with different speed, to find your S-Values for up and down.
[image]

oraherb, 2019-10-06:
@svenhb bro i subscribe to your youtube  and you make a dvd plotter thats what im doing now  and thanks it works but the servo is traveling to far like 180degrees   can u give me the exact setting of your dvd mini plotter ?

svenhb, 2019-10-06:
My dvd plotter uses stepper motor for Z, but the egg-bot I use a servo: 100 for pen up, 400 for pen down

oraherb, 2019-10-06:
@svenhb bro it works but  how can i setup my mini pen plotter  its x and y axis travel too far my range is only 40mm   even i configure in grbl setting??   im using  DRV8825 step driver   do i have to connect the M0 M1 M2 to +5v  like a4988 to have a mircro stepping? if ot how can i make this mini cnc  draw  small image??
[image]

svenhb, 2019-10-06:
If you move by 10 mm via virtual joystick, does it travel 10 mm? Then it's ok.
You can resize your drawing, depending on source a) SVG during conversion, b) via menu 'GCode transform. If you have an image you need to define the size on the tab 'image correction'.
Most probably your grbl settings are not correct. I set M0,1,2 al to +5V
GRBL-Settings:
$100=107.000
$101=107.000
$102=107.000
$110=10000.000
$111=10000.000
$112=10000.000
$120=500.000
$121=500.000
$122=500.000
See also http://svenhb.bplaced.net/?CNC___Mini_CNC_Plotter

oraherb, 2019-10-07:
@svenhb bro can u give me idea how to make an gcode  that will fit in  my mini cnc from dvd i mean  i tried everything but the gcode i created always exceed  the limit of my mini cnc 30mm  my stepper x and y always running out of axis

svenhb, 2019-10-07:
Does your x axis move by 5 mm if you send x5 ? Or if you click on '5' in virtual joystick?

oraherb, 2019-10-07:
X-axis: Engine = 400 = steps / revolution = 1,111 steps / degree 
setting the Y parameters: 
$ 100 = 1,111; 
$ 110 = 10000 (max. Speed - tried); 
$ 120 = 500 (max. Acceleration -ausprobiert)
Y-axis: Engine = 400 = steps / revolution + reduction 1:50 = 20,000 steps / revolution = 55.555 steps / degree 
setting the X-parameters: 
$ 101 = 55,555; 
$ 111 = 800 (max. Speed - tried); 
$ 121 = 500 (max acceleration -. Tried)

oraherb, 2019-10-07:
i tried those setting and it move 9mm

oraherb, 2019-10-07:
when i tried this your setting theng press 5 in virtual joystick it move 1mm per jog

$100=107.000
$101=107.000
$102=107.000
$110=10000.000
$111=10000.000
$112=10000.000
$120=500.000
$121=500.000
$122=500.000

svenhb, 2019-10-07:
Aha. You may change $100 to 1.111/9*5 = 0.617 and $101 to 30.86.... (or the other way arround *9/5)
Makes no sense to change any graphics scaling when your hardware reacts wrong!
BTW: did you uncheck "STOP Jogging on mouse up" - then on click causes one complete move for e.g. 5 mm.

oraherb, 2019-10-07:
@svenhb hey bro  when ipress the Y axis it also move the X axis in virtual  joystick but when i press the X      it doesnt move the way what u think is an hardware?

svenhb, 2019-10-07:
Check the COM window to what was sent to grbl. 
Are you using the hardware which is mentioned in your grbl-link  (https://github.com/bdring/Grbl_Pen_Servo) ? Perhaps some signal-wires are mixed up?

> Default setup.
> **The default setup is for a buildlog.net** Pen/Laser Bot Controller. If you are not using that, you need to make sure your schematic matches the pin defined in cpu_map.h

oraherb, 2019-10-07:
yes bro im using that grbl

svenhb, 2019-10-07:
If you move 5 mm in UGS, does your hardware move by 5 mm?
I think you need to play with your $100 / $101 setup.

oraherb, 2019-10-07:
yes bro it accurately move 5mm if i move 5mm in ugs

oraherb, 2019-10-07:
@svenhb hey bro help me with this how can i generate and gcode image that size is 10mm height and  10mm width  i dont know how i can resize this image to make it fit with my minimum axis [image]

svenhb, 2019-10-07:
Go to the tab 'Image correction' and play with 'width' and 'height'

oraherb, 2019-10-07:
ok bro thanks

winder, 2019-10-07:
@oraherb I'm glad that you're getting help, but please move this discussion on the right project page -> https://github.com/svenhb/GRBL-Plotter

svenhb, 2019-10-07:
@winder can you close the thread here?
@oraherb go on here: https://github.com/svenhb/GRBL-Plotter/issues/new/choose

closed 2019-10-07 by winder

oraherb, 2019-10-07:
ok  guys thanks