winder/Universal-G-Code-Sender #1281: 0x9e in console when using Spindle override button

mike-pittelko, 2019-10-06:
## Problem description 
Pressing the "spindle" button in the overrides panel doesn't enable or disable the spindle, but does enter 0x9e in the console

### Expected Behavior
Spindle to turn on, or off, depending on it's state.

### Actual Behavior
see 0x9e in console, and spindle does not change state.

### Steps to Reproduce the Problem
  1. Open the program
  2. Select the override panel
  3. Have the console open
  4. Press the "spindle" button
 5. 0x9e appears in console, no M command, and nothing happens.

### Specifications
#### Version

  UGS Platform 2.0 - Nightly Build  ~Oct 1 2019

#### Operating system
Both windows 10, and UGS on Raspian

#### Platform

  GRBL 1.1f

breiler, 2019-10-23:
I agree, this is a bit unintuitive.

The byte 0x9e is the correct command to send as its a real time command sent to GRBL which should be handled outside its planner buffer (unlike M-commands). 

But it should only be available when in HOLD-state, i.e. when paused. 
Does anyone use this and maybe explain its use case?

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands:
> 0x9E : Toggle Spindle Stop
>
> Toggles spindle enable or disable state immediately, but only while in the HOLD state.
> The command is otherwise ignored, especially while in motion. This prevents accidental disabling during a job that can either destroy the part/machine or cause personal injury. Industrial machines handle the spindle stop override similarly.
> When motion restarts via cycle start, the last spindle state will be restored and wait 4.0 seconds (configurable) before resuming the tool path. This ensures the user doesn't forget to turn it back on.
> While disabled, spindle speed override values may still be altered and will be in effect once the spindle is re-enabled.
> If a safety door is opened, the DOOR state will supersede the spindle stop override, where it will manage the spindle re-energizing itself upon closing the door and resuming. The prior spindle stop override state is cleared and reset.

lightingnerd, 2019-12-01:
I ended up here as well, however, the problem I am experiencing also occurs during G-Code execution. I have tried the nightly platform build, the 2.0 platform build, and even the previous 1.09 classic build, and the spindle won't turn on. However, when I tried executing a M3 S1000 via the command line it worked perfectly. 

Anyway, thank you guys for working on this project it's looking great! Hopefully the bug can be located...